Blueforcer/awtrix3 #172: [FEATURE REQUEST] - Allow to setup alarms via APIs

aleesposito85, 2023-06-06:
# Feature Request

### Is your feature request related to a problem?

Allow to setup alarms via APIs

aleesposito85, 2023-06-06:
> You can set a timer which plays a custom sound when the configured time has expired.

That is cool, but doesn't have the same functionalities, like: snooze and recurrence. Can also be a bit tedius doing the math :)

TheGoblinHero, 2023-06-20:
Good idea. I'd like this to be implemented too.

Blueforcer, 2023-07-20:
timer and alarms will probably be removed in the next versions. Everyone currently has their own idea and function wishes that can all be created via the smarthome.
On the one hand this follows the basic idea that awtrix is a smarthome companion, on the other hand it is so much more flexible.
Even snooze can be implemented with it.

burkasaurusrex, 2023-07-20:
Yea, makes sense. The only pushback would be for reliability. For something like an alarm, it would be great to have as few points of failure as possible. So some API functionality to schedule the next alarm so the alarm goes off even without WiFi, smarthome, power, etc would be great in my mind. But I understand it’s generally redundant when connected to a smarthome.

Blueforcer, 2023-07-20:
But tying the functions to reliability, when awtrix lives only from the smart home and its connection anyway, is not a decisive argument in my opinion. if you have a power failure, or connection loss, then you can no longer set the alarm via API.

Blueforcer, 2023-07-26:
closing since timer will be removed in the next version

closed 2023-07-26 by Blueforcer