Blueforcer/awtrix3 #174: [BUG] Playing sound is not ok

mcuiobroker, 2023-06-06:
# Bug report

## Describe the bug
I tried to play a RTTTL sound, but it is more one tone as a melodie. It seems to be one tone and in the background with little volume a melodie?
[image]
Sample is from 
https://www.laub-home.de/wiki/RTTTL_Songs 

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [0.68 ]

## To Reproduce

Steps to reproduce the behavior:
MQTT -Entry 
[image]
[image]

## Expected behavior

A clear and concise description of what you expected to happen.
If I let it play with https://adamonsoon.github.io/rtttl-play/ on a PC its sounds normal like a melodie.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-06-06:
It's not a software bug. The build in buzzer is crap. You can replace it with a better one, but nothing to do here from software side

closed 2023-06-06 by Blueforcer

mcuiobroker, 2023-06-06:
How can I update the buzzer?

mcuiobroker, 2023-06-06:
Is the anywhere a description how to open it?

Blueforcer, 2023-06-06:
On my discord server:

https://discord.com/channels/546407049148366859/1101071217986900049

And

https://discord.com/channels/546407049148366859/1100377495146082314

mcuiobroker, 2023-06-06:
Jetzt habe ich die Anleitung. Danke.