Blueforcer/awtrix3 #175: [QUESTION] Who wants a "half-dead" Ulanzi TC001 for free?

hagenbuch, 2023-06-07:
# Question

I've got a new Ulanzi TC001 that "bricked" while flashing awtrix-light. Whichever way I tried, I don't get it back to run. Ulanzi will send me a replacement, but maybe the old hardware is of some interest for one of you? Maybe it runs again if you resolder the flash chip? Maybe you'll find the physical reason for the bricking as several users seem to report this problem? (I'm having a second device which flashed fine to awtrix-light when I used to "upload" method..)

I know this has nothing to do with awtrix-light, I just would like to send the device instead of trashing it.

I would send it for free inside Germany, maybe other European countries too.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## Logs

None

## Additional context

None

Blueforcer, 2023-06-07:
I could use it vor further delvelopment :)
Komme aus Frankfurt

hagenbuch, 2023-06-07:
Kannst mir Deine Adresse gerne an post at dellekom dot de mailen...

hagenbuch, 2023-08-13:
Maybe not yet: An underdischarged Lithium battery might be dead to the point it can not be charged anymore, - thus it consumes all the current and gets hot (beware: Can start a fire!)

That could explain why the device does not see enough current to start by itself. Replace the battery first!