Blueforcer/awtrix3 #178: [BUG] Apps state after restart always true

BenAhrdt, 2023-06-08:
if i deactivetd some native app like "bat" and i restart the tc001 the app will be displayed.
is it possible to store the activation state?

Blueforcer, 2023-06-08:
It is stored.
How you deactivate it?

marlan99, 2023-06-08:
i recognized the same issue but when i start the tc001 when the battery was empty and i plugged in the cable again.
maybe a similar behaviour ?

marlan99, 2023-06-08:
maybe its only the activation state of the app time which will not be stored?

Blueforcer, 2023-06-08:
If someone would answering my question, there would be no need for guessing 😉

BenAhrdt, 2023-06-08:
I do it with the ioBroke Adapter.
I think the Adapter did it with
/api/apps/ with the paylod {name: appname, shown: false}
Or something… it works, until you restart the tc001

klein0r, 2023-06-08:
> If you still want to use the app API you need to send it after each reboot. It's only temporary

Thanks - I will remove that feature (since it's deprecated)

Blueforcer, 2023-06-08:
> > If you still want to use the app API you need to send it after each reboot. It's only temporary
> 
> Thanks - I will remove that feature (since it's deprecated)

If possible just uncomment it. In the future I try to add it again in a slightly different way. Because the current state cause more questions than it helps

closed 2023-06-08 by BenAhrdt