Blueforcer/awtrix3 #181: D1 Mini (China) nicht kompatibel?[QUESTION]

Elektroniker0486, 2023-06-08:
Ist die D1 Mini aus China mit der Awtrix2_uprgrade.bin nicht kompatibel. Im onlineflasher kommt die Meldung, das die Hardware nicht kompatibel sei. Wenn ich mit dem ESP Homeflasher die awtrix2_upgrade.bin flashe dann gibt es kein Lebenszeichen.
Dieser D1 hat mit der alten Awtrix Version einwandfrei funktioniert.

Wird das noch gefixt?

Vielen Dank im voraus!

Add here your question.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-06-08:
ESP8266 ist nicht kompatibel
Wenn du die alte Hardware hast, kannst du den ESP8266 D1 Mini 1:1 gegen den  ESP32 D1 Mini austauschen.

closed 2023-06-08 by Blueforcer

Elektroniker0486, 2023-06-08:
Ja, ich habe die alte Hardware mit deiner Awtrix Platine. Super, vielen Dank für die Info.