Blueforcer/awtrix3 #182: [FEATURE REQUEST] Emoji Support

blaxer, 2023-06-08:
# Feature Request

### Is your feature request related to a problem?

No

### Describe the solution / feature you'd like

I know this might be a superfluous request but it would be cool if emoji could be supported in the notification text and or as an icon. 

### Describe alternatives you've considered

Right now I have nodered flows that extract emoji from incoming notification text and replaces them with icons. It's a bit of a faff and means I have to upload and synchronize icons amongst clocks. 

### Additional context

This is obviously just a nice to have kind of thing.

Blueforcer, 2023-06-08:
How is this supposed to work?
It's only possible to show text or jpg/gifs

blaxer, 2023-06-08:
Ahh okay, well had to ask right? 😀

Blueforcer, 2023-06-08:
> Ahh okay, well had to ask right? 😀 Thank you for the consideration.

Sure, but unfortunately there is no other way. Uploading the icons you want is the easiest and customizable solution.