Blueforcer/awtrix3 #184: [QUESTION]Komma wird eine Zeile zu hoch gesetzt? Soll das so?

mcuiobroker, 2023-06-08:
# Question

Soll das Komma so dargestellt werden? Geht es nicht anders?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [0.68 ]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

[image]

Blueforcer, 2023-06-08:
Du warst doch grad noch im discord aktiv 😅

Die Schriftart wurde absichtlich so gemacht.

mcuiobroker, 2023-06-08:
Sorry, switche hin und herr. War gerade auf github

mcuiobroker, 2023-06-08:
Kann man da grundsätzlich eine andere Schriftart einstellen?

Blueforcer, 2023-06-08:
Nein das geht nicht. Die Schriftart wurde speziell für Awtrix entwickelt.

mcuiobroker, 2023-06-08:
Ok. Danke.

closed 2023-06-08 by mcuiobroker