Blueforcer/awtrix3 #185: [BUG]

marlan99, 2023-06-09:
i recognized a behaviour while running on battery.
i have two devices (FW v0.68).

when the battery shows approx. 50% (+-2%) SoC of the battery it automatically turns off.
you can turn it on but after a few minutes without charging it will again be turned off again.

above 55% there is no such behaviour.

Maybe the calculation is somehow wrong or just the device does not give the appropriate information ?

Blueforcer, 2023-06-09:
The calculation is as close as possible, and works great in 99%.
No one knows how exactly the analog value wich can be read needs to be mapped to percentage value. 
The range was done by measuring the raw min value and max. You can check the raw value in the Status API.

But if you have a broken battery wich cannot deliver the current at lower voltage anymore, the clock will switch off I think.

closed 2023-06-09 by Blueforcer

sansonifabio, 2023-06-19:
Also confirm. When the Battery below 55%, the clock can then no longer be reached via the API, the adapter then turns yellow and the clock has to be switched on again manually, but then switches itself off again after a short time.
**Is it possible to set this function with the battery so that it only happens at say 30%?**
Greetings Fabio

Auch bestätige. Wenn die Batterie unter 55% ist das die Uhr dann über die API nicht mehr erreichbar ist, der Adapter geht dann auf gelb und die Uhr muss manuell wieder angeschaltet werden, schaltet sich aber dann nach kurzer Zeit wieder aus.
**Ist es möglich diese Funktion mit der Batterie so ein zustellen das es erts bei sagen wir 30% passiert?**
Grüße Fabio

Blueforcer, 2023-09-18:
Awtrix does nothing with the battery percentage. Just display.
You can set the mapping via via dev.json.