Blueforcer/awtrix3 #186: delete or set disable installed apps

arteck, 2023-06-10:
please make some apps (eyes, date, clock, batt) switchable off direct after reboot

Blueforcer, 2023-06-10:
Go into your onscreen menu -> Apps and choose what apps you want, or don't want.

closed 2023-06-10 by Blueforcer

RK62, 2023-06-18:
How can I permanently disable the installed apps on a flashed Awtrix2? 
Currently I always send a mqtt.0.*.apps a 
`[{name: "time", show: true}, {name: "hum", show: false}, {name: "temp", show: false }] `
but the setting is lost after a reboot.

Blueforcer, 2023-06-18:
This API is not documented for a good reason 😅

Use this
https://blueforcer.github.io/awtrix-light/#/onscreen

RK62, 2023-06-18:
How to use onscreen when my awtrix2 has no buttons?

Blueforcer, 2023-06-18:
If you're limited in hardware, you're currently limited in software Features.
For now use the /app API and send it once as retained message. Zlthen it will be applied on each start automatically

pacmac, 2023-11-21:
I can't get this to work at all, I have published to the mqtt to `awtrix_1234xxx/apps` but the display just continues to show all of the built in apps, what am I doing wrong ??

```
awtrix_1234xxx/apps

[
{"name":"date","pos":0,"show":false},
{"name":"temp","pos":2,"show":false},
{"name":"hum","show":false,"pos":0},
{"name":"bat","show":false}
]
```