Blueforcer/awtrix3 #187: [BUG] Cannot update 0.67 to 0.68

waywit, 2023-06-11:
# Bug report

## Describe the bug

Updating from the Ulanzi clock running 0.67 over the menu result in downloading 0% - 100% and then restarting with 0.67.
Updating over the WebUI with the provided Firmware .bin file from the release page does the same... it looks like updating, but during the restart and the WebUI is still showing 0.67.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.67 on the clock, update to 0.68

## To Reproduce

Steps to reproduce the behavior:
- Update in clock menu
- Update with firmware file

## Expected behavior

Updating firmware works and the clock shows 0.68 during boot up and in WebUI.

Thanks for the great work :-)

pklijzing, 2023-06-11:
From the release notes:

Note: due an mistake this version will show 0.67. But its contains all this changes. This will be fixed with the next version

waywit, 2023-06-11:
Oh thanks… I missed this one 😊

closed 2023-06-11 by Blueforcer