Blueforcer/awtrix3 #188: [BUG] - v0.69 Colored fragments causes reboot

geekofweek, 2023-06-12:
# Bug report

## Describe the bug

When sending a payload with colored fragments on v0.69 it will cause the device to reboot.  I was able to verify this on two different devices and testing various different payloads with colored fragments being the only one I noticed it to crash on.  Rolling back to v0.68 solved the issue

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## To Reproduce

Steps to reproduce the behavior:

Sending this payload will cause the device to reboot

          { 
          "icon": "icon_name",
          "pushIcon": 0,
          "repeat": 2,
          "stack": false,
          "text": [
            {
              "t": "Red ",
              "c": "FF0000"
            },
            {
              "t": "White",
              "c": "FFFFFF"
            }
          ]
          }

## Expected behavior

Not crash and reboot when sending said payload

Blueforcer, 2023-06-12:
> I am not using color fragments but I am seeing my clocks crash / reboot when sending notifications on v0.69. Have not figured out the formula yet though I will do some more testing.

Send me your JSON so I can reproduce it

Blueforcer, 2023-06-12:
@geekofweek Please try this one:
[firmware.zip](https://github.com/Blueforcer/awtrix-light/files/11727688/firmware.zip)

geekofweek, 2023-06-12:
@Blueforcer that seems to fix it.  Tested with both notification and app custom app.  I'll let it run some and report back if I see anything

Pocky0013, 2023-06-13:
it was the same problem for me. Reboot when sending notification. The new version solves the problem, but the communication with the MQTT server is really super slow. What I didn't have with the old version

Blueforcer, 2023-06-13:
> But the communication with the MQTT server is really super slow. What I didn't have with the old version.
> 
> I receive the information from the buttons without delay. But receiving the notification on the display takes about 5-10 seconds before being displayed

Please open another issue for that.

geekofweek, 2023-06-13:
@Blueforcer the original issue seems to be resolved, it's been about a day and haven't had it crash on both devices.

closed 2023-06-14 by Blueforcer