Blueforcer/awtrix3 #189: [BUG] mqtt Einstellungen defekt?

Beast-Panik, 2023-06-12:
# Bug report

## Describe the bug

Ich habe mehrere Uhren in Netzwerk. Bei einer sind aber die MQTT Einstellungen weg bzw  die Eingabe ist wir auf den Screenshots zu sehen nicht möglich, nur Host wird angezeigt. Bei einer anderen Uhr ist alles wie gewohnt. Cache vom Browser löschen, anderer Browser etc. hat nicht geholfen. Wo liegt der Fehler?

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.69
war auch schon zwei Versionen zu or so, habe Updates abgewartet da ich dachte es wird dann gefixt aber anscheinend gab es das problem noch nicht. Habe hier zumindest auch nichts via suche gefunden.

## Screenshots
[image]
[image]

Blueforcer, 2023-06-12:
Hast du was in der Config.json händisch geändert?
Eine andere Ursache ist mir nicht bekannt und auch noch keinen Report diesbezüglich gehabt.

Beast-Panik, 2023-06-12:
Nein, da Haber ich nichts geändert. Wüsste auch nicht wie. Habe Grad aber einmal neu geflasht weil der reset Button nicht funktionierte, jetzt scheint die Konfiguration wieder ok. Sorry das ich nicht eher an ein neu Flaschen gedacht habe, kam erst drauf als ich weiter nach dem Fehler gegoogelt habe 🥴

Blueforcer, 2023-06-12:
Dann schließe ich hier erstmal. kannste gern wieder aufmachen wenn du den fehler evtl eingrenzen kannst oder nochmal bekommst

Beast-Panik, 2023-06-12:
Ich habe nach einer reset möglichkeit gesucht, auch beim alten awtrix, dann habe ich bei irgend einem Fehler gelesen das er nach dem neu flashen weg war. So kam ich drauf. Zu awtrix light direkt findet man tatsächlich nur das hier. Habe noch zwei verpackte Uhren hier, sollte das bei einer der beiden nochmal auftauchen, melde ich nicht. Die letzten Wochen war erst Mal der Umzug von iobroker zu Homeassistent Thema, primär angeregt durch die Uhren mit denen ich bei der mqtt Umsetzung in iobroker irgendwann zu sehr genervt war 😀🤷
Erstmal danke!

huncksas, 2023-11-18:
Hi,
bei mir ist gleiches passiert.
Ich hatte zunächst MQTT eingerichtet, alles lief soweit und dann habe ich mich jedoch dazu entschlossen lieber "direkt" über die API die Ulanzi Uhr anzusteuern - also habe ich die MQTT einstellungen gelöscht, gespeichert und neu gestartet.
Seitdem sieht bei mir die MQTT-Seite so aus wie oben angezeigt.
Ich hab allerdings die awtrix_light 0.90 installiert.

Beste Grüße

Blueforcer, 2023-11-18:
Manchmal scheint es das der ESP es nicht schafft die JSON fertig zu schreiben, warum auch immer. Vlt reboot zu schnell getriggert? Muss ich mal testen

huncksas, 2023-11-18:
Die json sieht für mich recht vollständig aus, beginnt mit { und endet mit }...
MQTT Abschnitt innerhalb der json sieht wie folgt aus:

> ...
>   "Secondary DNS": "1.1.1.1",
>   "param-box8": "MQTT",
>   "Broker": "",
>   "Port": null,
>   "Username": "",
>   "Password": "",
>   "Prefix": "awtrix_one",
>   "Homeassistant Discovery": false,
>   "param-box15": "Time",
> ...

Mir fällt allerdings gerade auch auf, dass dadurch das Menü anders ist - der Punkt für ICONS z.b. fehlt auch:
[image]

Blueforcer, 2023-11-18:
Ich muss mal schauen wie das funktioniert. Nutze für das Webinterface auch nur eine fertige Lib.
Denke mal er erstellt erst die JSON und sortiert nach einander ein.