Blueforcer/awtrix3 #191: [FEATURE REQUEST] - Minimum Brightness Level (Trim)

geekofweek, 2023-06-12:
# Feature Request

### Is your feature request related to a problem?

When the brightness level is set to the minimum, around 1%, I notice some flickering to off with certain colors.  It will expose itself using rainbow and various colors.  Almost gives it a blinking effect.  Now in reality this isn't a huge ordeal if you manually set the brightness level up to 5% and the issue goes away.  With Auto Brightness it can dim down to 1%, I prefer to keep them as dim as is needed but manual over ride solves it to an extent.  

### Describe the solution / feature you'd like

Expose a setting / feature, not so hidden or not, to set the minimum brightness level the matrix will go to (trim).

### Describe alternatives you've considered

 Problem can be solved using an external light level sensor and adjusting with external automations.

Chrischan-git, 2023-06-29:
Hi! Can anyone tell me the unit for the min_brightness/max_brightness setting over the dev.json file?
Since they are 2/180 per default, it cannot be a percentage, right?
ThX!

Blueforcer, 2023-06-30:
It's the internal brightness value wich is used by the matrix. It's from 0-255 so there is no unit for that. You can use Byte 😜

Chrischan-git, 2023-06-30:
Thanks a lot, that helps.
The range was unknown (maybe you can include this in the docu). This way I can derive the approx percentage value, as the setting on the display is in the percentage range...
And Byte is perfectly fine ;-)