Blueforcer/awtrix3 #194: Notification in "Buffer mode"

Pocky0013, 2023-06-14:
in version 0.62, when I sent a first notification with a duration time and then sent a second notification, the first notification was directly disabled and displayed the second notification.

Now the first notification disappears only when the duration time is over and the second notification arrives (like a buffer mode)

it's annoying because, for example, to set the time of a wake-up call, I used the notifications in HA automation loops to update the value of my timers.

Is there a way to overwrite the current notifications directly when a second notification arrives?

Many Thank's

Blueforcer, 2023-06-14:
You should read the changelogs and documentation before opening a "bug" rebort :)

set "stack" to false
https://blueforcer.github.io/awtrix-light/#/api?id=json-properties

closed 2023-06-14 by Blueforcer

Pocky0013, 2023-06-14:
I'm sorry for the inconvenience. As I had problems yesterday with the coloring of the text 
(solved by the firmware you sent), I'm not wary that it could come from elsewhere. You're doing a great job and it's really cool to make all of this available.