Blueforcer/awtrix3 #197: [BUG] Clock does not reconnect to Wi-Fi after it was disabled overnight

dimw, 2023-06-18:
# Bug report

## Describe the bug

My Wi-Fi - to which the clock is connected - is configured to turn off automatically during the night. Every morning, when the Wi-Fi is on again, the clock does not reconnect. However, it reconnects to Wi-Fi properly when the clock is turned off and on again. Disabling and enabling Wi-Fi again does not lead clock to reconnect. 

This issue is not present when Wi-Fi disappears for a short time (several minutes). In this case the clock reconnects properly.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.70
  - Wi-Fi router

## To Reproduce

Steps to reproduce the behavior:
1. Connect the clock to a Wi-Fi.
2. Turn Wi-Fi off for ~6 hours.
3. Turn Wi-Fi on.
4. **Issue:** The clock does not reconnect to Wi-Fi.

## Expected behavior

The clock reconnects to Wi-Fi shortly after the signal gets available again.

## Logs

These are the logs I see in the morning on the clock while Wi-Fi is available for at least 2h already.

```text
[124394730][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[124399729][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[124400127] [setCurrentApp]: Publish current app temp-outside-forecast
[124404252][E][vfs_api.cpp:105] open(): /littlefs/alarms.json does not exist, no permits for creation
[124404730][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[124409527] [setCurrentApp]: Publish current app precipitation-forecast
[124409729][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[124414252][E][vfs_api.cpp:105] open(): /littlefs/alarms.json does not exist, no permits for creation
[124414730][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
```

## Additional context

Kudos for the awesome project!

mslookup, 2023-06-28:
I can confirm the bug, i have the same situation and my Ulanzi Pixel clock does also not reconnect automatically.

Blueforcer, 2023-06-29:
Seems like a bug in the arduino framework.
autoreconnect is enabled and worked for me.
https://github.com/Blueforcer/awtrix-light/blob/a96ce12366584bc1b423093d2206654754052252/lib/webserver/esp-fs-webserver.cpp#L172

dont know what else to change..

davlaw, 2023-07-06:
Well, feeling kinda like what the heck??   Checked my wifi logs and showing over 35 hours on most 2.4 devices of connectivity and having both of my ulanzi devices fall offline again this morning, an experiment is in order!  

Gonna fall back versions for a bit. Was at .70 falling back to version .69

Lets see?!?

Blueforcer, 2023-07-06:
since the alpha version, there have been no changes to the wifi functions.
So downgrade will not solve it, for sure

davlaw, 2023-07-07:
Understand,  and agree, but had to try something.

Just a note, clock visually appeared to be working normally, it was cycling thru it screens.  What I noticed some values were not as I was expecting such as temp/humidity and weather.

Entities in Home Assistant were no long updated, shown "unavailable" , both clocks

Unable to log into clock web interface.   Both clocks

Power off and back on restored operation on both clocks and their web interface and conx to HA

Interesting note, both fell offline at about 4am my time, odd.  So I and looking at some outside influence something on my network.

What effect does mqtt or loss of connection have if any?

cebrac, 2023-07-09:
Can confirm that this occasionally happens to my clock (0.70) even when WiFi doesn't turn off (maybe just drops out for a second?)

Blueforcer, 2023-07-10:
Some users reports that ehys have problems with some Ubiquiti routers.
They had to disable some optimizing features and sekect a static wifi channel
https://discord.com/channels/546407049148366859/1124762310133698713

mslookup, 2023-07-10:
Interesting! I have indeed Ubiquiti UniFi WiFi APs in place. Will try it out!

davlaw, 2023-07-10:
Asus RT-AX58U/merlin here.  I have fallen back to .69 for now with some device monitoring on the 2 ulanzi devices.  

I did catch and disable Roaming assistant, it can disconnect when below certain signal strength, I split the 2.4 and 5 bands too.  Did this sometime ago, could see this causing all kind of problems with devices getting all whacked out getting random disconnects and such. Did not see much value in that option.

Blueforcer, 2023-07-11:
> If anybody could please copy the info for non Discord users, that'd be much appreciated <3

"Also i disabled nightly channel optimization on my unifi controller"

"I'm finding little hints in a google search that some ESP32s have problems on wifi channel 11 -- people posting logs that say channel 11 then that they changed their channel to auto and the problem resolved itself"

davlaw, 2023-07-14:
I know, you said it would not make any difference, but I tried, .69 would not reconnect after a ~ 1 hour power failure this morning.

Battery stayed up  tho!

Blueforcer, 2023-07-27:
any news here?

mslookup, 2023-07-27:
With disabling overnight WiFi channel optimization in UniFi Controller, I got an uptime of ~2 weeks. For me, it’s not a problem any more.

cebrac, 2023-07-27:
Same here. Shame I had to disable that feature but if it's an issue with the ESP32 and not the specific firmware, understandable that it can't be fixed.

Blueforcer, 2023-07-27:
Sorry

closed 2023-07-27 by Blueforcer

zanaca, 2023-07-28:
Hi guys!

I had the same problem here and I do not have any UniFi Controller.  I have and old router (2.4ghz only) running OpenWrt. I had to turn off the router for a while and then I have noticed the TC001 was not connected to the wifi. Had to restart it.

hardoverflow, 2023-08-10:
I can confirm that two installations of the clock have from time to time the same disconnection issues as discribed here. Only a reboot helps. But sometimes, i need to reboot twice. And yes, all the clocks depends in UniFi APs.

quantum-fc101, 2024-04-27:
Same issues here using Ruckus Unleashed AP's clock disconnects after awhile. Clock is connected to an IOT VLAN with 2.4G only and Background Scanning is switched off.