Blueforcer/awtrix3 #202: [BUG?] Can't connect to Wifi

Calimerorulez, 2023-06-23:
I've flashed 0.70, reset the clock, and now the clock is in AP Mode.
When trying to connect to Wifi, available networks aren't shown, and the following error is shown looping in the terminal log (via the flasher website):

```
E (165468) wifi:Set status to INIT
E (165515) wifi:Set status to INIT
E (165598) wifi:Set status to INIT
E (165688) wifi:Set status to INIT
E (165730) wifi:Set status to INIT
E (165828) wifi:Set status to INIT
E (165865) wifi:Set status to INIT
E (165956) wifi:Set status to INIT
E (166040) wifi:Set status to INIT
```

What am I doing wrong?

floschibo, 2023-06-25:
Same for me. Worked fine for 10 days. Suddenly No WiFi connection.

Calimerorulez, 2023-06-25:
I’ve found some kind of workaround. Using esprit, I’ve fully erased the ESP32 in the clock. Reflashing it via the site with the Awtrix firmware gave the same error at first, but after restarting the clock several times, it now connects.

But it’s still a hit or miss like it feels.

Calimerorulez, 2023-06-26:
Well, it was fun while it lasted. This morning it won't connect anymore. Even after restarting the clock.

floschibo, 2023-06-26:
It's working again. For now there is no issue for me. 🫢🤷

Blueforcer, 2023-06-29:
Does the SSID that you are using relate to a 2.4GHz or 5GHz network?
The ESP32 can only connect to 2.4GHz.
If you’re using a dual band router which uses a single SSID for both bands then try turning-off 5GHZ support or rename one of it.

cebrac, 2023-07-11:
I also have a Unifi setup. Since I am not sure which features to turn off (for lack of Discord) I turned most optimization / roaming related ones off for the IoT network. Hopefully this will help.

Regarding 5 GHz network, I think that since the ESP* don't see it, it shouldn't be relevant whether there is one?

mimimi0815, 2024-07-19:
I have the same issue on two sepaertated clock synchron. Both running well for ~ half a month. Then suddenly both can't connect to wlan anymore after nightly wlan off i  the morning. Both stays in ap mode. It is also hard to connect in this mode. It often not work. Then i could Connect after several tries to one of them and look to the config. It looks good and everything is correct (files,mqtt,..) except wlan. There is no connection show. Scan of wlan does not work, there is nothing shown when i click. And when i try a manual connection with right ssid and passwort only "error" is shown. I am using the newest awtrix version (0.96) on ulanzi tc001. Please help. Thanks