Blueforcer/awtrix3 #203: Prevent clock from entering AP Mode

cebrac, 2023-06-26:
# Feature Request

I have a pretty stable Wifi but sometimes upon resetting the clock(s), some of them apparently have trouble connecting on the first try, which causes them to enter (and remain stuck in) AP Mode.

It'd be very helpful if there was a setting to disable this behaviour after initial setup.

As a failsafe against locking oneself out, the setting could be available via the onscreen menu, be disabled by holding a certain button on startup, or by restarting several times in a row or just disable itself after a longer period of being unable to connect to a Wifi.

Thank you for considering this.

hytrek, 2023-07-10:
Same here.. I do not know how to leave this mode except reset

Blueforcer, 2023-07-10:
just restart the clock. There is a timeout of 10 or 15 seconds.

closed 2023-07-26 by Blueforcer

cebrac, 2023-08-05:
Since this is still an ongoing and very annoying problem for me, I'd be willing to make another donation if it would help find the time to implement this. Can discuss details if interested.

sladexos, 2023-08-25:
Same Problem here. I think the problem is my Mesh-Network with 3AP. So I see the same network three times in the found wifis.

So I can't use the clock...

cebrac, 2023-08-26:
Since the clock runs more stable now with the current release, I have deacticated the scheduled reboots as they do no longer seem to be necessary (with very few exceptions).

So for me personally, I can live with how it is now.

@sladexos this may help:
https://github.com/Blueforcer/awtrix-light/issues/197#issuecomment-1628301707

wildpash, 2024-01-05:
Really strange, after the watch was only powered by the battery and it switched off by itself after a few hours, it now falls into AP MODE after a restart.
I have a Frizbox router and the clock can be reached in the Firtzbox's WiFi despite the AP Mode display. I can access the web interface (IP from Fritzbox) and unfortunately the watch remains in AP MODE even after a restart.

topperharley80, 2024-02-13:
> Really strange, after the clock was only powered by the battery and it switched off by itself after a few hours, it now falls into AP MODE after a restart. I have a Frizbox router and the clock can be reached in the Firtzbox's WiFi despite the AP Mode display. I can access the web interface (IP from Fritzbox) and unfortunately the watch remains in AP MODE even after a restart.
> 
> **Edit:** Restarting the clock using both buttons was unsuccessful, even after several attempts. However, a "restart ESP" via the web interface under the "Network" tab worked.
> 
> I have not made any changes to the WIFI credentials or stored a static IP in the web interface.
> 
> What I noticed was that I could see the AP in the nearby WLAN networks, even though the clock had an IP from the Fritzbox and I could therefore also reach the web interface.
> 
> PS: I didn't try to connect to the AP.

Exact the same issue. Fritzbox and AP Mode.

I could fix it with a permantent IP adress and a restart ESP und network