Blueforcer/awtrix3 #209: [BUG] clock back to 01.01.1070

BenAhrdt, 2023-07-05:
Meine Uhr war heute morgen aus (hängt am USB) und nach dem Einschalten wieder auf dem Datum 01.0.1970

Blueforcer, 2023-07-05:
Dann konnte sich die Uhr nicht mit dem NTP server verbinden um die Uhrzeit zu holen.
Entweder falsche Server eingetragen, kein Internet oder der NTP Server zur Zeit nicht erreichbar

BenAhrdt, 2023-07-05:
ABer die Uhr funktionierte schon 4 Wochen

Blueforcer, 2023-07-05:
Dann ist es auch unwahrscheinlich daß sich die Software in der Zeit ändert.

Wie gesagt, die in meinem letzten Kommentar angesprochenen Punkte liegen am nächsten.

BenAhrdt, 2023-07-05:
Wie kann ich jetzt wieder an die Uhr kommen? Sie zeigt die IP An, aber ich komme nicht drauf.

Blueforcer, 2023-07-05:
Welche IP zeigt sie denn an?

BenAhrdt, 2023-07-05:
192.168.2.41    --- DIe habe ich auch kofiguriert und war auch aktuell.
Bei update erscheint ein error und in der Fritzbox wird das Gerät nicht angezeigt.
Also es scheint sich nicht mit dem WLAN zu verbinden, obwohl eine IP angezeigt wird.
Bis gestern abend war die Uhr normal im WLAN.
Nun kann ich sie nicht mehr erreichen

Blueforcer, 2023-07-05:
wenn du die IP manuell als statisch Konfiguriert hast, dann wird die natürlich angezeigt. Wenn du da einen Fehler gemacht hast, wirst du nicht mehr drauf kommen. Hilft nur noch neu flashen.

BenAhrdt, 2023-07-05:
die IP stimmt ja. wie gesagt lief es ja 4 wochen bis heute morgen.

Blueforcer, 2023-07-05:
Hast du ne statische IP im Webinterface angegeben?
Wenn ja dann bitte zurücksetzen und mit ner dynamischen probieren.

BenAhrdt, 2023-07-05:
Funktioniert wieder.
Neu gelasht, aber scheinbar war das WLAN abgeschmiert. Da das Gerät als einziges im WLAN hing, viel es nicht auf.
Könnte man noch ein not Connected als Hinweis geben?

closed 2023-07-05 by BenAhrdt