Blueforcer/awtrix3 #212: [FEATURE REQUEST] Automatic calculation of Notification duration for scroll text

febalci, 2023-07-12:
### Describe the solution / feature you'd like

If the notification text is too long as scrolling text, the duration needs to be calculated according to the scroll speed in order for all text to be shown. I use newsapi.org to get latest headlines in a python script, but need to also calculate the duration separately as seconds, based on character length of the text,  to be sent to awtrix light. 

I use it like this:
```
newstext="The news text i get from the newsapi.org"
newstextlength = len(newstext)
duration = int(newstextlength / 6.5) + 1
```
Of course this is only for the default scrollspeed. Is there a way to incorporate "duration" of the text scroll to be automatically calculated? Of course the manual duration set as it is now should also be kept for other purposes.

### Additional context

Thanks for the great work...

febalci, 2023-07-13:
Thanks for the solution, works perfectly...

closed 2023-07-13 by febalci

bachoo786, 2023-11-24:
> Thanks for the solution, works perfectly...

i would love to see news on my ulanzi time display. can you please share your setup in depth please? thanks.

febalci, 2023-11-25:
> hey @febalci i would love to see news on my ulanzi time display. can you please share your setup in depth please? thanks.

1. Go to [newsapi.org](https://newsapi.org/). Create an API key. 
2. Create news.py python file with this, changing the 'YOUR_NEWSAPI_API_KEY' with your API key and 'YOUR_COUNTRY_CODE ' with country code, like 'us' for 'USA'.
```
#!/usr/local/bin/python3
import json
import urllib3
import requests

newsapiKey = 'YOUR_NEWSAPI_API_KEY'
newspage = 'https://newsapi.org/v2/top-headlines?country=YOUR_COUNTRY_CODE&pageSize=2&apiKey=' + newsapiKey

result_json = ''
urllib3.disable_warnings()
try:
    response = requests.get(url=newspage,verify=False)
except requests.exceptions.RequestException as e:
    print('error')
    quit()

JSON_response = response.json()

result_json = JSON_response['articles'][0]['title'] + ' / ' + JSON_response['articles'][1]['title']

print (result_json.upper())
```
4. Test you news.py with `python3 news.py` and make sure it works.
5. The following lines are for homeassistant configuration; if you are not using homeassistant you can maybe use mqtt.
6. Create a command line sensor in your homeassistant configuration.yaml: 
```
command_line:
  - sensor:
      name: AWTRIX News
      command: "python3 external_data/news.py"
      scan_interval: 900
      json_attributes:
        - news
        - secs
      value_template: '{{ value_json }}'
```
7. Create an automation:
```
- alias: AWTRIX Ulanzi News
  trigger:
    - platform: state
      entity_id: sensor.awtrix_news
  action:
    - service: mqtt.publish
      data:
        qos: 2
        retain: true
        payload: >-
          {
            "text": "{{ state_attr('sensor.awtrix_news','news') }}",
            "icon": "news",
            "rainbow": false,
            "repeat": 1,
            "pushIcon": 1
          }
        topic: awtrix_ulanzi/custom/news
```

febalci, 2023-11-27:
Hey @bachoo786 , the python file works fine, i also tested yours. Assuming you already restarted homeassistant to read command_line config from config.yaml; there may be one or both of these 2 causes: 

1. Problem with the Read-write-execute permissions of /srv/docker/homeassistant/python_scripts/news.py folder or file.
2. Homeassistant keeps its own python modules separate from the system. Although you install json, urllib3 and requests python modules on the linux of homeassistant, that is not installed in the docker instance of homeassistant. I do not remember it well, but i guess you can find info on internet how to install python modules in docker instance of homeassistant.

febalci, 2023-11-27:
@bachoo786 ,

my bad. Sorry for this, i pasted the wrong and outdated python file previously. Here is the correct one:

```
#!/usr/local/bin/python3
import json
import urllib3
import requests

newsapiKey = 'YOUR_NEWSAPI_API_KEY'
newspage = 'https://newsapi.org/v2/top-headlines?country=YOUR_COUNTRY_CODE&pageSize=2&apiKey=' + newsapiKey

textSpeed = 6.5

result_json = ''
urllib3.disable_warnings()
try:
    response = requests.get(url=newspage,verify=False)
except requests.exceptions.RequestException as e:
    print("SSL Error")
    quit()

JSON_response = response.json()

result_json = JSON_response['articles'][0]['title'] + ' / ' + JSON_response['articles'][1]['title']

print ('{ "news":"'+result_json.upper()+'", "secs":'+str(int(len(result_json) / textSpeed)+1)+'}')
```

bachoo786, 2023-11-27:
> @bachoo786 ,
>
> my bad. Sorry for this, i pasted the wrong and outdated python file previously. Here is the correct one:
>
> ```
> #!/usr/local/bin/python3
> import json
> import urllib3
> import requests [...]

it finally worked :) thank you