Blueforcer/awtrix3 #216: [BUG]

bicmac, 2023-07-14:
I have 2 custom Apps setups over IOBroker AWTRIX-Light Adapter in my TC001.
I have hidden all internal apps accept data and time.
auto Switch for apps is enabled and works if i dont have setupd Cusom apps. As soon as i add a custum app he will ignore the autoswitch and walk through the cycle exactly once. That he will stopp and not start from beginning.
As the IOBroker Adapter use the official API i think it is maybe a bug in AWTRIX-Light FW. But i have no clue how to debug that.
Actual FW is installed on the Device. Is there any way to downgrade to a older Version as i think it was working allready once.

bicmac, 2023-07-19:
Hier ein paar mehr Infos aus dem IPBroker AWTRIX Light Ticket:
https://github.com/klein0r/ioBroker.awtrix-light/issues/39

Es scheint so das es einen Fehle rgibt sobald man Textscrolling disabled in Custoum Apps und der Text länger ist.
Zitat: "Es muss an dem Text Scrolling liegen. Ich habe das gleiche Problem.
Solange der Text nicht zu lang ist, funktioniert die automatische App Rotation. Sobald der Text aber zu lang ist, der Haken "noScroll" gesetzt ist, verharrt das Display bei der App. Erst wenn der Text wieder aufs Display passt geht's weiter."

@Blueforcer kannst Du das bei Dir nachstellen oder hast Du ne Idee warum die Apps nicht mehr weiterlaufen sondern einfach stehen bleiben.

bicmac, 2023-07-19:
Hier vorsichtshalber nochmal der komplette Logoutput für den Request aus dem IOBroker:

`2023-07-19 21:51:23.691  - debug: awtrix-light.0 (12107) sending "POST" request to "/api/custom?name=aussen" with data: {"background":"#000000","text":"21,8 °C","textCase":2,"repeat":1,"color":"#FFFFFF","noScroll":true,"icon":"40921"}
2023-07-19 21:51:23.808  - debug: awtrix-light.0 (12107) received 200 response from "/api/custom?name=aussen" with content: "OK"`

Blueforcer, 2023-07-19:
wie ich mir dachte:

Sobald du "repeat" nutzt, wird die App nicht mehr nach einer Zeit beendet sondern dann wenn der Text X mal durchgescrollt ist.
Mit noScroll verhindert du es aber und es wird nie eintreten.

Ich könnte das evtl noch abfangen das bei noScroll wieder auf Zeit geschaut wird und nicht auf repeat.

Bis dahin kannst du repeat entfernen oder auf 0 setzen

bicmac, 2023-07-19:
danke. Muss dazu warten bis Matthias seinen IOBroker Adapter anpasst. Denn momentan kann ich 0 nicht setzen für das Repeat über den Adapter. Ich hab ihm aber schon geschrieben ob er es anpassen kann.
Ggf findest Du ja auch Zeit es relativ zeitnah bei Dir irgendwie abzufangen.

Blueforcer, 2023-07-19:
Dann ist das aber generell falsch.
Repeat ist optional in awteix und sollte nicht Standardmäßig auf 1 sei.

In awteix ist es Standard 0 wenn's nicht angegeben wird

Blueforcer, 2023-07-25:
Sorry fehler von mir. Grad nochmal geschaut, -1 ist standard. Denn bei 0 ist er fertig mit scrollen

closed 2023-07-25 by Blueforcer

klein0r, 2023-07-26:
Also liegt es an `repeat: 1` dass nicht zur nächsten App gewechselt wird? Oder verstehe ich `noScroll` falsch?

Blueforcer, 2023-08-03:
Ja. `noScroll` verhindert das `repeat` runterzählt. Wenn repeat angegeben ist, wird diese variable um bei jedem kompletten durchscrollen um eins verringert. Wenn 0 dann ist die App fertig.
ist `repeat `= -1 (wenns z.b nicht in der json angegeben wird) dann wird `repeat ` nicht beachtet sondern die `duration `bzw globale apptime.
In der obrigen Konstellation war es nun aber so das Awtrix gewartet hat bis `repeat `<1 ist, was aber nie passiert ist weil das scrollen mit `noScroll` verhindert wurde

klein0r, 2023-08-03:
Perfekt, danke für die Erklärung

PR #233 (2023-07-25): 0.71
### 📈 New Features
- Allows to loop the sound in notifications sou youre able to build your own timer or Alarm Clock for example. closes #232 
- Youre able to save Customapps wich are reloaded automatically after boot
- Adds effects wich can be used in Notifications, CustomApps and as a general background layer.
- Allows to set the Home Assistant MQTT Discovery prefix. Closes #215 

### 🐞 Bug Fixes and Optimizations
- Sets comma one pixel down. closes #208
- Removes Eyes native app and add it as an effect
- Removes Timer and Alarm, not in code for now, but in documentation. This gives you [... cut]
commit d7501fa2 (2023-07-25): V0.71

### 📈 New Features
- Allows to loop the sound in notifications sou youre able to build your own timer or Alarm Clock for example. closes #232 
- Youre able to save Customapps wich are reloaded automatically after boot
- Adds effects wich can be used in Notifications, CustomApps and as a general background layer.
- Allows to set the Home Assistant MQTT Discovery prefix. Closes #215 

### 🐞 Bug Fixes and Optimizations
- Sets comma one pixel down. closes #208
- Removes Eyes native app and add it as an effect
- Removes Timer and Alarm, not in code for now, but in documentation. This gives y [... cut]