Blueforcer/awtrix3 #219: Display problems with some icons [BUG]

Chrischan-git, 2023-07-17:
Hi. I'm having some trouble with some of the icons from the LaMetric database.
You can e.g. test the Icon ID: 54298, this is one I made myself, but I have the same problems with some others.
This is an 8-bit icon. If I use the preview when uploading, everything is ok, the animation is also displayed correctly in the file manager.
But then there are display problems on the device (red background during animation).

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.7

Steps to reproduce the behavior:
Upload the icon to the device and use it for a notification or app

Blueforcer, 2023-07-17:
Try to not use transparency. fill this pixels with Black

Chrischan-git, 2023-07-17:
Jep, this works!
Unfortunately, this cannot be done in the LaMetric editor, even if I put black in the free fields there.
So you have to upload the file, download via filebrowser, convert it with the online tool und reupload in the filebrowser...a bit laborious ;-)

But thanks for the solution.

closed 2023-07-19 by Blueforcer