Blueforcer/awtrix3 #220: [FEATURE REQUEST]

blutack, 2023-07-18:
# Feature Request

### Describe the solution / feature you'd like

It would be great if awtrix included a subset of the WLED style 2D animation effects. Ideally, this would be implemented such that notifications etc would replace the effect, then go back to it once the notification time has elapsed - same as the built in clock app.

### Describe alternatives you've considered

Try and run the animations via streaming either via artnet or a custom app: the high frame rates involved would likely result in either jank or crashing, and I'm not sure how it would interact with notifications.

Keep using WLED: but then I don't get the nice custom apps and notification features of awtrix!

### Additional context

I currently have a ulanzi flashed with WLED running various 2D matrix effects.

[image]

I would like to use Awtrix but I don't want to lose my screensaver type effects and be looking at a clock all day.

I am happy to do the work myself, but I'm not sure where it fits with your vision of awtrix. I could jam it into the moodlight feature, but I suspect it would work better as a built in app (Effects?) so that it fits nicely into the existing UI/usage paradigms.
I suggest I start by putting the scaffold in place to add the extra built in app, then porting one or two WLED 2D effects across. Does that sound sensible?

blutack, 2023-07-20:
Closing as we've moved the discussion to the PR

closed 2023-07-20 by blutack

Blueforcer, 2023-07-22:
@blutack Have a look at discord if you have time