Blueforcer/awtrix3 #223: [BUG]

Dinges28, 2023-07-19:
# Bug report

## Describe the bug

From the moment a indicator is "Blinking", and an automatic transform to the next app happens, the blinking becomes erratically or the indicator will turn black (although its status stays "on") 

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.7

## To Reproduce

turn on (one of the 3, or all if you like ;-)) an indicator in blinking mode <"blink": true>
wait till the automatic transition
indicator starts flashing erratically 

(sometimes it also happens immediately , the moment you turn on the indicator in blinking mode

## Expected behavior

previous firmware (0.6) had 2 indicators, that would continuous blink in a slow pattern when turned on in blinking mode

## Screenshots

https://github.com/Blueforcer/awtrix-light/assets/39988228/23e802f6-3ab9-4d82-b33b-a9dd3a2c3cc3

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-07-19:
You should not use a boolean value for the blink key ...

Dinges28, 2023-07-19:
> You should not use a boolean value for the blink key ...

I've used this previous. What should it be than??

```
          - service: mqtt.publish
            data:
              qos: 0
              retain: false
              topic: "{{my_display}}/indicator3"
              payload: |-
                {   
                  "color": [255, 0, 255],
                  "blink": true
                }
```

Blueforcer, 2023-07-19:
According to the documentation, the blinking interval.

https://blueforcer.github.io/awtrix-light/#/api?id=colored-indicators

Dinges28, 2023-07-19:
@Blueforcer Thanx, that works like a charm indeed. 
Did not know it had to be formatted like this, the previous was create based on the blueprint of smarthome junkie.

closed 2023-07-19 by Dinges28