Blueforcer/awtrix3 #224: [BUG?] Clock shuts down when matrix is turned off for 2h

TheGroundZero, 2023-07-19:
# Bug report

## Describe the bug

I'm not entirely certain whether this is a bug or expected behaviour.

I have an automation that turns off the display (and the transitions) at night.
I noticed in the morning, the device would be off and I'd need to manually turn it on again by holding the 2 buttons.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.70

## To Reproduce

Here's the automation:

```yaml
automation:
  - id: awtrix_night
    alias: Awtrix turn off at night
    description: Turn the display of the Awtrix off at night
    trigger:
      - type: illuminance
        platform: device
        device_id: 7cb35d836fcd914d47337990435ba877
        entity_id: 1b9cec4097d58a857a0dbcca27e86cc8
        domain: sensor
        above: 20
        for:
          seconds: 30
        id: "on"
      - type: illuminance
        platform: device
        device_id: 7cb35d836fcd914d47337990435ba877
        entity_id: 1b9cec4097d58a857a0dbcca27e86cc8
        domain: sensor
        below: 20
        for:
          seconds: 30
        id: "off"
    condition: []
    action:
      - choose:
          - conditions:
              - alias: Turn on
                condition: trigger
                id: "on"
            sequence:
              - alias: Awtrix Matrix on
                type: turn_on
                device_id: 7cb35d836fcd914d47337990435ba877
                entity_id: light.awtrix_dd2824_matrix
                domain: light
              - alias: Transition on
                type: turn_on
                device_id: 7cb35d836fcd914d47337990435ba877
                entity_id: switch.awtrix_dd2824_transition
                domain: switch
          - conditions:
              - alias: Turn off
                condition: trigger
                id: "off"
            sequence:
              - alias: Awtrix Matrix off
                type: turn_off
                device_id: 7cb35d836fcd914d47337990435ba877
                entity_id: light.awtrix_dd2824_matrix
                domain: light
              - alias: Transition off
                type: turn_off
                device_id: 7cb35d836fcd914d47337990435ba877
                entity_id: switch.awtrix_dd2824_transition
                domain: switch
    mode: single
```

## Expected behavior

The device's screen turns off at night and on during the day.

## Screenshots

[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

blutack, 2023-07-19:
Are you sure your HA automation is actually working? There's no record in that Logbook screenshot you posted of the "turn_on" ever being triggered. Try to trigger the "turn on" action manually from the device page in HA?

Your illuminance trigger value might be too high - mine is showing ~17 on my desk with a reasonable amount of daylight.
Check the HA history for the illuminance sensor maybe?

TheGroundZero, 2023-07-19:
My assumption is that the device completely powered off, as the lux sensor is also unavaible, thus the 'on' trigger cannot occur.

I just left the office at noon and turned off the lights, dropping the lux value below my trigger.
Returning to the office and turning the lights on again, did turn the clock on this time.
[image]

Current illuminance is 45lux, btw.
I did use sunrise/sunset before. But I kinda like the idea of it not being on when not needed. Also, might cause issues in winter.

Blueforcer, 2023-07-19:
Awtrix is unable to physically turn on or off the clock.

If you need to turn it on with holding both buttons, the clock was off and the ESP had no power.

Sounds like battery drain.

TheGroundZero, 2023-07-19:
How long can the clock last on battery?
It's constantly powered over USB while I'm at my desk, but I do flick the switch on my power block when I leave for the night.

Blueforcer, 2023-07-19:
According to Ulanzi 5h max