Blueforcer/awtrix3 #226: [BUG] Settings are not stored

sotima, 2023-07-20:
# Bug report

## Describe the bug

When sending either a MQTT message to the topic **awtrix/settings** with a payload '{"TSPEED":750}' the setting TSPEED still remains at 400. Also other settings do not have any effect (e.g. '{"TFORMAT":"%H:%M"}' does not change the time format of my clock. Same behaviour when sent via curl to the [IP]/api/settings url.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.70
  - Home Assistant 2023.7 with HA Discovery enabled. MQTT-Prefix set to awtrix/

## To Reproduce

Steps to reproduce the behavior:
- In HA use the service MQTT publish to the topic awtrix/settings with a payload '{"TFORMAT":"%H:%M"}' or
- Using linux send via curl:
`curl --header "Content-Type: application/json"   --request POST   --data '{"TFORMAT":"%H%M"}'   http://192.168.3.23/api/settings`

## Expected behavior

A clear and concise description of what you expected to happen.
The changed settings should immediately show on the clock.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-07-20:
i believe its not a software bug, because its used by many users without issues.
Awtrix saves all to a json:
https://github.com/Blueforcer/awtrix-light/blob/main/src/Globals.cpp#L193

Have a look at the serial console while saving, and make sure you have enough free flash memory (You can check this in the filemanager top right)

sotima, 2023-07-20:
> i believe its not a software bug, because its used by many users without issues. Awtrix saves all settings: https://github.com/Blueforcer/awtrix-light/blob/main/src/Globals.cpp#L193
> 
> Have a look at the serial console while saving and post the output.

Yes, it seems you are right! Tested again, using Node-Red instead of the service-call in HA and there it works as described! Have to look into my HA-Installation..

closed 2023-07-20 by sotima