Blueforcer/awtrix3 #227: [FEATURE REQUEST] Incorporate on-board peripherals in custom apps or notifications

cskwrd, 2023-07-21:
# Feature Request

I am always frustrated when I have to send information the device already has.

### Describe the solution / feature you'd like

I would be nice if the sensor and time/date data were available for use in custom apps.

### Describe alternatives you've considered

Sending the data from external source like a smart home hub.

### Additional context
It could be that the docs are unclear, and I don't understand the difference between a custom app and a notification. It seems the only difference is a custom app persists and an notification does not.

udo1toni, 2023-07-21:
A notification has the ability to play a sound. You can push the button at the ulanzi to acknowledge the notification. See https://blueforcer.github.io/awtrix-light/#/api?id=json-properties for details.

Blueforcer, 2023-07-21:
I dont see any sense to use the date and time from a native clock, to use it in customapp.. why not using the native app?
Sending time and date from a smarthome system is easy a.f.

According to your other question:
Its like your smartphone. you have fixed apps and like the name says one time notifications wich stays on top of the apps.

closed 2023-07-27 by Blueforcer