Blueforcer/awtrix3 #229: Compatibility with WROOM32 and connection schema

Gobolin80, 2023-07-21:
Hi, I'm trying to make the project using a WROOM32 like this: https://www.amazon.it/SP-Cow-Sviluppo-ESP-WROOM-32-ESP-32S-Bluetooth/dp/B08CCYWZN3

I can flash the firmware without problems, but after entering the wifi SID and password it returns "ERROR!" (and nothing else). 
So I ask you if the device is compatible or if it works only with" ESP32 D1 mini".
I take this opportunity to ask you if the wiring diagram of the connections is available and specifically an indication on how to connect the LED matrix. From the code it seems to me that it should be connected to the GPIO15, but I'm not sure.

Thanks too much

Blueforcer, 2023-07-21:
It's only tested with a ESP32 D1 mini and nodemcu.
But Ulanzi also have a ESP32-WROOM-32D

Pinout:

<table>
		<thead>
			<tr>
				<th>ESP32 PIN Number</th>
				<th>GPIO Name</th>
				<th>Usage or part</th>
				<th>PixelIt Mapping</th>
			</tr>
		</thead>
			<tbody>
			</tr>
				<tr>
				<td>6</td>
				<td>ADC6 / GPIO34</td>
				<td>Battery sensor (displays battery % on wake)</td>
				<td>n/a</td>
			</tr>
			</tr>
				<tr>
				<td>7</td>
				<td>ADC7 / GPIO35</td>
				<td>LDR (light) sensor (GL5516)</td>
				<td>n/a</td>
			</tr>
			<tr>
				<td>8</td>
				<td>GPIO32 </td>
				<td>Matrix Type 2 / Row Major</td>
				<td>n/a</td>
			</tr>
			<tr>
				<td>11</td>
				<td>GPIO26</td>
				<td>Left button</td>
				<td>Pin_D4 (Active low)</td>
			</tr>
			<tr>
				<td>12</td>
				<td>GPIO27</td>
				<td>Middle button</td>
				<td>Pin_D5 (Active low)</td>
			</tr>
			<tr>
				<td>13</td>
				<td>GPIO14</td>
				<td>Right button</td>
				<td>Pin_D6 (Active low)</td>
			</tr>
			<tr>
				<td>23</td>
				<td>GPIO15</td>
				<td>Buzzer</td>
				<td>n/a</td>
			</tr>
			<tr>
				<td>n/a</td>
				<td>n/a</td>
				<td>Temperature and Humidity Sensors (SHT3x)</td>
				<td>I2C address 0x44</td>
			</tr>			
		</tbody>
</table>

Blueforcer, 2023-07-21:
Ulanzi uses a WROOM-32D. Beside this only D1 Mini and some NodeMCU  are tested from other users:

<table>
		<thead>
			<tr>
				<th>GPIO</th>
				<th>Usage or part</th>
			</tr>
		</thead>
			<tbody>
			</tr>
				<tr>
				<td>GPIO34</td>
				<td>Battery sensor</td>
			</tr>
			</tr>
				<tr>
				<td>GPIO35</td>
				<td>LDR (light) sensor (GL5516)</td>
			</tr>
			<tr>
				<td>GPIO32</td>
				<td>Matrix</td>
			</tr>
			<tr>
				<td>GPIO26</td>
				<td>Left button</td>
			</tr>
			<tr>
				<td>GPIO27</td>
				<td>Middle button</td>
			</tr>
			<tr>
				<td>GPIO14</td>
				<td>Right button</td>
			</tr>
			<tr>
				<td>GPIO15</td>
				<td>Buzzer</td>
			</tr>
			<tr>
				<td>GPIO 21/22</td>
				<td>Temperature and Humidity Sensors (SHT3x)</td>
			</tr>			
		</tbody>
</table>

Gobolin80, 2023-07-21:
Thank you so much for your fast response!
After a lot of retry my WROOM32 works! I don't know why, but suddenly it has connected to the wifi.
I take advantage of your kindness for one last question: 
I tried to connect this 8x32 LED matrix that I already owned (https://it.aliexpress.com/item/4000759456651.html?Spm=A2G0o.order_detail.order_detail_item.3.42ef582fybvwhg&gatewayadapt=glo2ita) but all leds are off :( 
Do you know any LED matrix that works?

closed 2023-07-21 by Blueforcer

Gobolin80, 2023-07-22:
Ok,
in this thread I found the solution of the problem: https://discord.com/channels/546407049148366859/1119531762456993852/1119561218768785468

The correct output pin for Matix Led is **GPIO21** and is necessary to set custom property **"matrix": 2** in the **dev.json* file (https://blueforcer.github.io/awtrix-light/#/dev?id=json-properties).
Now it works!

I hope that this post can help other users.

Blueforcer, 2023-07-22:
You're referring to the  upgrade firmware for old AWTRIX2 Mainboard . Not ulanzi.
You should say wich firmware you're using ;)