Blueforcer/awtrix3 #236: Interface not working using reverse proxy (HTTPS and/or subdirectory)

wilfredktlr, 2023-07-26:
# Bug report

## Describe the bug

I use a reverse proxy (HAProxy) to access the webinterface. When Awtrix want's to load a resource (eg. `http://${window.location.hostname}/[filename]`) it fails because the browser cannot access it when the webinterface is accessed using HTTPS or a subdirectory. This causes the webinterface to not work correctly (eg. the menu).

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.71

## To Reproduce

Steps to reproduce the behavior:
- Setup a reverse proxy to access the site using HTTPS and/or a subdirectory.

## Expected behavior

No JavaScript errors and a working menu.

## Screenshots

An example of a fixed protocol and path.

[image]

The JavaScript error and non-working menu.

[image]

## Logs

N/A

## Additional context

A possible solution would be to make the URL relative (eg. `var e=new URL('./config.json');`).

Blueforcer, 2023-07-26:
This "bug" should belong to the webinterface library. 
https://github.com/cotestatnt/esp-fs-webserver

My knowlege for that is limited. Feel free to open a PR

wilfredktlr, 2023-07-26:
Thanks. I didn't realize that this was an external library. I've opened an issue over there.

https://github.com/cotestatnt/esp-fs-webserver/issues/29

wilfredktlr, 2023-08-11:
The author of the esp-fs-webserver just fixed the issue should be fixed in version [1.2.3](https://github.com/cotestatnt/esp-fs-webserver/releases/tag/1.2.3).

https://github.com/cotestatnt/esp-fs-webserver/issues/29#issuecomment-1674354045

commit 05ed3cad (2023-08-18): 0.75

AWTRIX Light app for iOS is now in open beta test [See discord](https://discord.com/channels/546407049148366859/591248067735322624/1142077243145855036)

- Implements logarithmic gamma control for better colors. This gives more saturation in normal brightness and while in the past youre loosing pixels completely in low brightness, the new mechanism tries to lower the gamma in that lower range. You doesnt get the colors right in low brightness, but at least you see the pixel.
Best colors are at 30-60% brightness
- Decreases the Brightness range
- Awtrix webinterface now supports reverse pr [... cut]
closed 2023-08-18 by Blueforcer