Blueforcer/awtrix3 #239: [FEATURE REQUEST] Restart WiFi and MQTT when Wifi disconnected from Access Point

robercy, 2023-07-27:
# Feature Request
Try to restart WiFi and MQTT when Wifi disconnected from Access Point, caused by a power supply problem.

### Describe the solution / feature you'd like

Some kind of watchdog monitoring Wifi (!wifi.connected) and MQTT ( onMqttDisconnecte) connections, and after a X minutes of connection(s) down, restart mcu or try to reinitialize the connections.

This is a good starting point to think about: [https://esp32.com/viewtopic.php?t=3851](url).

Thanks.

Blueforcer, 2023-07-27:
Maybe duplicate of https://github.com/Blueforcer/awtrix-light/issues/197

reconnect already implemented and works great.
See
https://github.com/Blueforcer/awtrix-light/blob/a96ce12366584bc1b423093d2206654754052252/lib/webserver/esp-fs-webserver.cpp#L172

MQTT lib has build in reconnect.

robercy, 2023-07-27:
Uau !! Thats a too fast response!! Thank you.

Indeed, my apologies ! You are right ! My fault. 

Maybe I not pay attention in my test bed.

Thanks again.

closed 2023-07-27 by robercy