Blueforcer/awtrix3 #243: [BUG] Notification dismissing does not work sometimes

TheGoblinHero, 2023-08-01:
# Bug report

## Describe the bug

I've made an alarm clock on HA platform. It is triggered as notification with the following payload:
`{"text":"Wake!","icon":"alarmclock","effect":"Ripple","sound":"alarm1","duration":300,"loopSound":true}`
Works great, but sometimes I cannot dismiss it pressing the middle button on the clock. The bug is not stable, I cannot reproduce it, but I've already encountered it twice. This morning it happened again. Pressing\holding the middle button did nothing. After a minute or so I've tried to press the button again - and it dismissed.
The button itself is ok! I've checked the HA logs - it detects presses.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.71

Blueforcer, 2023-08-01:
Are you sure you sending the notification only once? If you send it multiple times it gets into the stack. If you dismiss you get the same notification instantly again.

TheGoblinHero, 2023-08-01:
Log looks ok:
```
1 August 2023
Button select turned off
07:51:20 - 5 hours ago
Button select turned on (worked third press)
07:51:20 - 5 hours ago
Button select turned off
07:50:08 - 5 hours ago
Button select turned on (does not work)
07:50:06 - 5 hours ago
Button select turned off
07:50:06 - 5 hours ago
Button select turned on  (does not work)
07:50:04 - 5 hours ago
Штора was opened triggered by automation Будильник №1 triggered by time set in Время будильника 1
07:50:02 - 5 hours ago
Будильник №1 triggered by time set in Время будильника 1   (main trigger)
07:50:00 - 5 hours ago - Traces
```
Is where a way to debug the presence of multiple notificatons?

Blueforcer, 2023-08-01:
no,
you can check your mqtt messages and quantity with MQTT Explorer:

i would also disable your automation and send it manually one time.

TheGoblinHero, 2023-08-02:
Tried to run the alarm action ~20 times manually - dismissed with the first try all of them :( As I've said the bug is not reproducible.  Will try to look into it next times it happens again.

Blueforcer, 2023-08-12:
any news here?

TheGoblinHero, 2023-08-14:
I was on vacation and returned only yesterday. Just updated to 0.73. Give me a week to test it.

TheGoblinHero, 2023-08-19:
Can't reproduce the bug anymore. Closing the ticket.

closed 2023-08-19 by TheGoblinHero