Blueforcer/awtrix3 #245: [FEATURE REQUEST] Making DFplayer clone proof

robercy, 2023-08-02:
Hello, I have news about this problem.

After unstable behavior, and a lot of research, I discover (as discussed on [ghmartin77/DFPlayerAnalyzer#10](https://github.com/ghmartin77/DFPlayerAnalyzer/issues/10) the final solution is changing the value of c_msSendSpace to 150 (line 365 on DFMiniMp2.h file), solving the problem, and I believe, will avoid future problems.

My suggestion, if possible and no cause other problem, will be change this value on next PR on github. Thanks.

closed 2023-08-03 by Blueforcer

robercy, 2023-08-03:
One more thing! 
The value of c_msSendSpace  (line 365 on DFMiniMp2.h file) is still the same, do not changed to 150 !

Blueforcer, 2023-08-03:
damn this commit wanst pushed xD
its updated now, and will be used in future updates, sorry

robercy, 2023-08-03:
Don't worry, my friend. Thats why you have ours eyes and attention. Great job.

PR #247 (2023-08-03): Dev
[image]

- ### 📈 New Features
- Adds 9 App transition effects wich can be set via settings API or new HA entity.
- Allows to change color and speed of the effects with "effectSettings".
- Allows to create custom color palettes for effects.
- Adds ScreenMirror page where you can download screenshots and GIFS of the current matrix content to share with the community and friends.
- Adds /api/screen endpoint to get the current matrix as an color array.
- Adds the possibility to ask for the color array via MQTT.
- Adds new modes to the time app wich can show a calendar in different ways next to [... cut]
commit 6c9f5c4f (2023-08-03): V0.72

[image]

- ### 📈 New Features
- Adds 9 App transition effects wich can be set via settings API or new HA entity.
- Allows to change color and speed of the effects with "effectSettings".
- Allows to create custom color palettes for effects.
- Adds ScreenMirror page where you can download screenshots and GIFS of the current matrix content to share with the community and friends.
- Adds /api/screen endpoint to get the current matrix as an color array.
- Adds the possibility to ask for the color array via MQTT.
- Adds new modes to the time app wich can show a calendar in different ways next [... cut]