Blueforcer/awtrix3 #254: [BUG] WLAN connection breaks off after a while

pr1m300, 2023-08-07:
# Bug report

## Describe the bug

After several hours, there is suddenly no connection to the WiFi and no current data from my custom apps are displayed, Awtrix's own apps continue to work.
The apps also continue to scroll normally and also display data.

After I restart the display, it works again for a few hours. In the meantime I have already reduced the custom apps to only 4. I transmit the data via MQTT every 30 seconds.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - AWTRIX-LIGHT version: 0.72

## To Reproduce

Steps to reproduce the behavior: I just have to wait long enough

Thanks

Blueforcer, 2023-08-07:
possible dublicate of https://github.com/Blueforcer/awtrix-light/issues/197

pr1m300, 2023-08-08:
Thank you for the link to the other thread.

I also have a UniFi system and have switched off the automatic channel search accordingly.

Unfortunately the problem persists, what could it be? I would be very grateful for any help or a thought-provoking impulse.

RDG88, 2023-08-08:
I had this issue also when my AP was on channel 11, i have Unifi's also. Try to change the channel for 2.4ghz to 1 or 6.

hytrek, 2023-08-08:
Same issue with me. Using Unifi as well. 
@RDG88 Sorry I am clueless on how to change wifi 2.4 to another channel...

Suggestion: force fw. of watch to re-try connecting to wifi when signal is lost. When signal is lost restart always helps for some undefined time.

robercy, 2023-08-08:
I possible fix, will be a watchdog: on wifi.disconect() start timer, if disconnected more than X minutes, then esp.restart().

Blueforcer, 2023-08-08:
Please read linked issue #197.
It's look like a hardware or Framework issue.
Wifi reconnect is already enabled 
https://github.com/Blueforcer/awtrix-light/blob/a96ce12366584bc1b423093d2206654754052252/lib/webserver/esp-fs-webserver.cpp#L172

robercy, 2023-08-09:
Thanks again. I read the issue #197, and the solution works indeed, but in few cases, didn't work. 
I agree with you, restarting can cause more trouble than help. I will look my setup, to find the problem.  Keep the great awtrix rolling.

Blog404DE, 2023-08-10:
> Thanks again. I read the issue #197, and the solution works indeed, but in few cases, didn't work. 

I ran into the same problem - also with Unifi hardware. I looked at the mentioned issue as well and completely disabled the "Channel Optimization". However, the problem persists - after a few hours (sometimes more, sometimes less) the wifi connection is lost again.

```
[5316279] [generateCustomPage]: PARSING FINISHED
[5320887] [setCurrentApp]: Publish current app historystromverbrauch
[5331543] [setCurrentApp]: Publish current app historyaussentemperatur
[5342199] [setCurrentApp]: Publish current app Time
[5352856] [setCurrentApp]: Publish current app aussentemperatur
[5360514] [setCurrentApp]: Publish current app luftfeuchtigkeit
[5371173] [setCurrentApp]: Publish current app wettertext
[5374264] [nextApp]: Switching to next app
[5374673] [setCurrentApp]: Publish current app stromverbrauch
[5376465] [parseCustomPage]: Single Page
[5376467] [generateCustomPage]: PARSING FINISHED
[5385346] [setCurrentApp]: Publish current app historystromverbrauch
[5396011] [setCurrentApp]: Publish current app historyaussentemperatur
[5406667] [setCurrentApp]: Publish current app Time
[5415473][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[5417345] [setCurrentApp]: Publish current app aussentemperatur
[5420472][E][WiFiClient.cpp:242] connect(): connect on fd 50, errno: 118, "Host is unreachable"
[5424745] [setCurrentApp]: Publish current app luftfeuchtigkeit
```

GitHoubi, 2023-08-10:
Same issues kept me busy the last to days. Everyting was fine on 0.71 and problems started with 0.72.

I could find a way around, now all is working stable again.

The issue seems not to be related to the FW itself but rather through the way of flashing it.

I flashed 0.72 over 0.71 using the integraded update function. 

Once I flashed the FW over the webinterface things worked again as expected.

robercy, 2023-08-11:
Kudos to @GitHoubi  !
I follow his guide, and wifi do not disconnect since applied.
Seems to be a flashing memory allocation problem.

GitHoubi, 2023-08-11:
I know it somewhat does not make sense since the FW certainly is exactly the same...

But everything seems to run more stable if flashed via Web UI instead of the button press update.

I tried so many times. 0.71, 0.72, 0.73 all of them get instable after a while when flashed via device directly.

robercy, 2023-08-11:
Thats right, @GitHoubi .
Like said a old man friend of mine:  There is more mystery inside a cpu/mcu than our vain philosophy dreams of.

GitHoubi, 2023-08-11:
> > Same issues kept me busy the last to days. Everyting was fine on 0.71 and problems started with 0.72.
> > I could find a way around, now all is working stable again.
> > The issue seems not to be related to the FW itself but rather through the way of flashing it.
> > I flashed 0.72 over 0.71 using the integraded update function.
> > Once I flashed the FW over the webinterface things worked again as expected.
> 
> with webinterface you mean the Awtrix light frontend and then the Update tab? Or do you mean connecting the device via usb cable to computer and then use the flasher (https://blueforcer.github.io/awtrix-light/#/flasher)?

correct, AWtrix light Frontend and the update tab. This seems to work fine, so does the WebFlasher on https://blueforcer.github.io/awtrix-light/#/flasher via USB. But the update function within the device itself, this seems to be part of the problem.

bimbolongo, 2023-08-11:
Unfortunately, none of the suggestions have worked for me so far.

What I've already tried.

Delete and flash again

Update from 0.71 to 0.72 or 0.73 via the web interface and with the online flasher.

Only a downgrade to 0.71 solves the problem and everything works again.

I am using a custom build with an ESP32 (ESP32 NodeMCU Module WLAN WiFi Dev Kit C)

If anyone has any suggestions to solve the problem I would be happy.

davlaw, 2023-08-11:
Maybe use tasmota method of using conx to mqtt?  If that fails reboot?

robercy, 2023-08-12:
In my projects, I use a state machine to keep wifi and/or mqtt alive.

pr1m300, 2023-08-12:
> Ich hatte dieses Problem auch, als mein AP auf Kanal 11 war, ich habe auch Unifis. Versuchen Sie, den Kanal für 2,4 GHz auf 1 oder 6 zu ändern.

For a few days now I have set all APs to channel 1. Since then I have had no more disconnections. Has been running for a little over 3 days now.
thanks for the tip

Blog404DE, 2023-08-12:
> For a few days now I have set all APs to channel 1. Since then I have had no more disconnections. Has been running for a little over 3 days now. thanks for the tip

interessting ... i had set my AP to channel 6 and i had disconnections. I've now changed it to channel 1 and "locked" my pixel clock to the AP near the clock. Maybe this will help ...

RoelG, 2023-08-12:
> interessting ... i had set my AP to channel 6 and i had disconnections. I've now changed it to channel 1 and "locked" my pixel clock to the AP near the clock. Maybe this will help ...

Hmmm this got me thinking. I configured my clock in my office. There it probably connected to my AP that's right there. Then I moved my clock to the living room. There I had once the issue that the clock lost connection after a while. Maybe because it switched to another AP in the living room. Since then I didn't have the issue anymore. Running fine now for 4 days.

hytrek, 2023-08-14:
Same here - using Unifi aswell-  after changing all my APs to channel 1 the clock stays online stable since 5 days. As I am not an expert for Wifi I have no clue weather this has some effects on other devices or the general performance.

Blueforcer, 2023-08-23:
seems like to be an hardware/framework issue after all of your reports.
So im closing here, unfortunally nothing i can do here.

closed 2023-08-23 by Blueforcer