Blueforcer/awtrix3 #258: [BUG]

CharlyMai, 2023-08-09:
# Bug report
Awtrix with  Ulanzi permanent reboot

## Describe the bug
   After install the 0.73 Version every thing looks fine.
   If i add the MQTT-Field (correctly) without the Home-Assistant Checkbox, the Ulanzi shows: 

     1) "Awtrix" in rainbow color 
     2) "Awtrix 10.10.30.150" in rainbow color with the correct IP
     3) "MQTT..." in rainbow color
     4) Time for a few 1/100 second
     5) "Check" in white color
     6) "no up :(" in white color
     7) "0.73" in rainbow color

And than it starts by No 1.......

## Additional information

- Devices involved:
     - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)

## To Reproduce

   Steps to reproduce the behavior:
     a) Firmware update
     b) connect to ioBroker (so far so good)
     c) enter MQTT IP and all other fields (permanent reboot)

BR
CharlyMai

Blueforcer, 2023-08-09:
Doesn't sound like youre using Homeassistant..?

Lampyridae1984, 2023-08-11:
> If youre using IOBroker

I have the same issue and I don't use IOBroker.

Blueforcer, 2023-08-11:
Then check with MQTT explorer wich messages arrive to awtrix.
If there is a message to /reboot you sent it to awtrix.

Lampyridae1984, 2023-08-11:
Fantastic! I can't even flash it manually with the old firmware because of the boot loop.

Blueforcer, 2023-08-11:
Read my comment and stop sending the reboot command. Easy.

You can still flash it via USB like any other ESP.

Lampyridae1984, 2023-08-11:
> You can still flash it via USB like any other ESP.

Well that's the thing, it keeps rebooting. I can neither see anything with MQTT explorer, nor does it stay connected to my PC. Every two seconds in reconnects to the PC via USB.

Blueforcer, 2023-08-11:
Wait, if it's reconnect to the PC, then it has nothing to do with the Firmware or ESP. the USB interface is separated and is always active even without a running ESP.

Are you sure your ulanzis battery is full of charge? It will not run with a empty battery even if it's connected via USB.

Lampyridae1984, 2023-08-11:
> Are you sure your ulanzis battery is full of charge? It will not run with a empty battery even if it's connected via USB.

Well, it was plugged in to an usb charger for 24h before I updated the firmware and I have only moved it to the PC för manual flashing. So as far as I know, the battery should be between 95-100%. The clock is not visible in windows or through the ESP Web Tool when it is turned off and connected through USB.

Lampyridae1984, 2023-08-11:
When I have it connected via USB and hold down both arrow buttons to make it start/boot, it basically connects and disconnects infinitely every 2 seconds - so flashing it manually seems out of the question.

Lampyridae1984, 2023-08-11:
Guess I have to fix the wireless connection to HA some how, thou not sure what is wrong. Everything worked fine with V0.71

As already mentioned, I'm using the Mosquito broker addon in HA - on port 1883. The only thing connected to it (except for the Ulanzi clock) is my Zigbee2mqtt network and three Shelly devices (three smart plugs and one outdoor thermometer).

Has anything changed in V.72 or V.73 that requires specifik settings? Do I need to reconfigure the clock somehow or add some settings to the Mosquito broker?

Lampyridae1984, 2023-08-11:
I'm tired, bloody and a bit perplexed... but it's finally working again. I had to disassemble it completely. After unplugging the battery and matrix, I could finally re-flash it. But while assembling it, the issue came back. Only after disassembling and assembling everything again roughly 5 times it "just" works.

I'm now literally scared s***less to flash or OTA any future updates. It's seems to be either poor soldering or imminent component failure, but I lack the equipment and know how to do anything about it. A multimeter doesn't really help me without proper soldering gear (and know-how). Ulanzi seriously needs to ramp up the quality control on this fairly simple ESP32 device. I've already pre-emptively ordered a new one, I hope the old ones holds until it arrives. 🤞

Blueforcer, 2023-08-21:
Since no more feedback comes from the issue creator, I close here. If there are still problems, please open again

closed 2023-08-21 by Blueforcer