Blueforcer/awtrix3 #262: [BUG]

mnul, 2023-08-15:
# Bug report

## Describe the bug

Cannot download any LaMetric icons through the web interface. Error "This ID doesn't exist". Preview works fine.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.74

## To Reproduce

Go to the webinterface of your device, switch to [icons]. enter any valid LaMetic Icon ID (e.g. 10400) and click preview, icon will show. 
If you cllick on download, error window pops up and says "This ID doesn't exist".

Blueforcer, 2023-08-15:
try another browser, make sure javascript is enabled, and clear the brosercache

mnul, 2023-08-18:
latest update seems to have fixed it ...

closed 2023-08-18 by mnul

Blueforcer, 2023-08-18:
mhh there was no change to this:)
but glad it work now

tebald1, 2024-05-12:
I'm facing the same issue. Preview works fine, download shows "this ID doesnt exist".

I'm running AWTRIX 3 0.96. Tried Firefox and Edge on Windows and Chrome on Android. Same error on all plattforms.

Any workaround for this issue?

imac941, 2024-05-16:
I have the same issue with version 0.96.

Does anyone have any ideas?