Blueforcer/awtrix3 #265: [BUG] Update error

Calimerorulez, 2023-08-19:
Running 0.74, I'm trying to update my Ulanzi clock. When clicking  'Start update' in Home Assistant, the display shows 'Err cnct'. I saw this behavior the first time in 0.73. Version 0.72 updated correctly to 0.73 when using this procedure.

GitHoubi, 2023-08-19:
> Running 0.74, I'm trying to update my Ulanzi clock. When clicking  'Start update' in Home Assistant, the display shows 'Err cnct'. I saw this behavior the first time in 0.73. Version 0.72 updated correctly to 0.73 when using this procedure.

Reboot the device and then try again to update fw immediately after the boot process finished. It might need a try or two but will eventually succeed to update the fw.

Blueforcer, 2023-08-19:
This is a bug in the http updater wich i cant change.
Sometimes it work sometimes not. no matter wich firmware.
its always better to upload the firmware to webinterface and will work everytime

closed 2023-08-19 by Blueforcer