Blueforcer/awtrix3 #271: Scroll repetitions are not applied [BUG]

FreunderNacht, 2023-08-23:
# Bug report

## Describe the bug

App repetitions sent via the html API are no longer applied.

The following is sent:

`sending "POST" request to "/api/custom?name=aussentemp" with data: {"text":"22,7 °C","textCase":2,"background":"#000000","color":"#ffeb72","repeat":4,"icon":"5836","duration":8,"rainbow":false}
sending "POST" request to "/api/custom?name=hausverbrauch" with data: {"text":"3.762 W","textCase":2,"background":"#000000","color":"#ffffff","repeat":4,"icon":"54708","duration":8}
sending "POST" request to "/api/custom?name=ladestandspeicher" with data: {"text":"62 %","textCase":2,"effect":"TwinklingStars","color":"#FFFFFF","repeat":4,"icon":"49521","duration":8}`

However, the text is scrolled only once and then jumps to the next app.

This is only the case since frimware version 0.81.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.81
  - IO-Broker Adapter version: v0.7.1

## To Reproduce

Steps to reproduce the behavior:

After updating the firmware, it is immediately noticeable.

The text scrolls only once and then jumps to the next app.

## Expected behavior

The set repetitions or the set time should be run through and only then the next app should be displayed.

See version 0.80.

## Screenshots

https://github.com/Blueforcer/awtrix-light/assets/69077006/9a58a638-68b5-4837-8478-fbf0d4dfef50

obrocke, 2023-08-23:
Same behavior for me. Apps are not repeating. Notifications are working with repeats. In version 0.80 everything is still fine.

Blueforcer, 2023-08-29:
should be fixed in v0.82

closed 2023-08-29 by Blueforcer