Blueforcer/awtrix3 #272: Infinit loop after setting MQTT credentials [BUG]

tommimaus, 2023-08-23:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: [ eg. v0.45 ]

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-08-23:
See pinned issue. https://github.com/Blueforcer/awtrix-light/issues/165
**Next time please fill out the issue template otherwise it will be closed immediately**

closed 2023-08-23 by Blueforcer

tommimaus, 2023-08-23:
> Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.
After setting the (correct) MQTT credentials and restarting the ULANZI it goes in an infinite loop .
The Order of DISPLAY Messages after Restart:
Rev. 0.81
AWTRIX
IP Address
MQTT ... 
short Display of Date Time
CHeck Upd
No Up :(

Restart the loop as written above.

The MQTT Data Transfer for has happened correctly.