Blueforcer/awtrix3 #274: [BUG] After connecting to WIFI and MQTT display stays dark

Benni620, 2023-08-25:
# Bug report

## Describe the bug

I have flashed a new Ulanzi via the web tool. Then connected to the WLAN as described in the instructions. 

Now the display does not seem to work correctly. When I start it, it shows the AWTRIX version. Then in the rainbow "AWTRIX" then the IP scrolls through. Then "MQTT".
And then for a fraction of a second the time is shown, then the display stays black.

It does not react to the MQTT commands, only "[PREFIX]/moodlight {"brightness":170, "color":[155,38,182]}" lets it light up briefly in the color.

I have already tried to reflash the SW. I am not sure if I made a mistake or if there is a HW defect.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - AWTRIX-LIGHT version: 0.81

## To Reproduce

It happens on every reboot

## Logs

```text
E (541) esp_core_dump_flash: No core du����ѥѥ���found!
E (541) esp_core_dump_flash: No core dump partition found!
[  5096][E][Preferences.cpp:483] getString(): nvs_get_str len fail: TFORMAT NOT_FOUND
[  5096][E][Preferences.cpp:483] getString(): nvs_get_str len fail: DFORMAT NOT_FOUND
[  5104][E][vfs_api.cpp:105] open(): /littlefs/dev.json does not exist, no permits for creation
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to [Wifiname]
```

About a little help I would be very grateful! If you need more information just ask.
Thank you

Blueforcer, 2023-08-25:
Please enable debug_mode
https://blueforcer.github.io/awtrix-light/#/dev

Benni620, 2023-08-25:
Ah yes exactly what I was looking for...

This is the log with debugging enabled. I sent a few MQTT commands and pressed the buttons.

```
E (541) esp_core_dump_flash: No core du����ѥѥ���found!
E (541) esp_core_dump_flash: No core dump partition found!
[  5105][E][Preferences.cpp:483] getString(): nvs_get_str len fail: TFORMAT NOT_FOUND
[  5106][E][Preferences.cpp:483] getString(): nvs_get_str len fail: DFORMAT NOT_FOUND
[5132] [setup]: Setup periphery
[5305] [loadSettings]: Webserver configuration loaded
[5305] [setMatrixLayout]: Set matrix layout to 0
Framebuffer_GFX::begin Width: 32 Height: 8 Num Pixels: 256
Connecting to [Wifiname]
..........................[13754] [setup]: My IP: 192.168.42.101
[17249] [setup]: Webserver loaded
[24733] [setup]: Starting Homeassistant discovery
[24737] [connect]: Connecting to MQTT with login
[29794] [onMqttConnected]: MQTT Connected
[29794] [onMqttConnected]: Subscribe to topic /brightness
[29827] [onMqttConnected]: Subscribe to topic /notify/dismiss
[29860] [onMqttConnected]: Subscribe to topic /notify
[29893] [onMqttConnected]: Subscribe to topic /custom/#
[29927] [onMqttConnected]: Subscribe to topic /switch
[29960] [onMqttConnected]: Subscribe to topic /settings
[29993] [onMqttConnected]: Subscribe to topic /previousapp
[30026] [onMqttConnected]: Subscribe to topic /nextapp
[30059] [onMqttConnected]: Subscribe to topic /doupdate
[30092] [onMqttConnected]: Subscribe to topic /nextapp
[30124] [onMqttConnected]: Subscribe to topic /apps
[30156] [onMqttConnected]: Subscribe to topic /power
[30187] [onMqttConnected]: Subscribe to topic /indicator1
[30218] [onMqttConnected]: Subscribe to topic /indicator2
[30249] [onMqttConnected]: Subscribe to topic /indicator3
[30280] [onMqttConnected]: Subscribe to topic /timeformat
[30311] [onMqttConnected]: Subscribe to topic /dateformat
[30342] [onMqttConnected]: Subscribe to topic /reboot
[30373] [onMqttConnected]: Subscribe to topic /moodlight
[30404] [onMqttConnected]: Subscribe to topic /sound
[30435] [onMqttConnected]: Subscribe to topic /sendscreen
[73045] [select_button_pressed]: Select button clicked
[75469] [right_button_pressed]: Right button clicked
[77244] [left_button_pressed]: Left button clicked
[87221] [onMqttMessage]: MQTT message received at topic awtrix/moodlight
[87222] [onMqttMessage]: Payload:  {"brightness":170,"color":[155,255,182]}  
[106280] [onMqttMessage]: MQTT message received at topic awtrix/indicator1
[106281] [onMqttMessage]: Payload:  {"color":[255,0,0]}
```

Blueforcer, 2023-08-25:
Mhh, I don't see any crash or reboot in the log. Is this all?!

Benni620, 2023-08-25:
Yes exactly, nothing happens after that. At least nothing else is displayed in the log

Blueforcer, 2023-08-25:
Battery is full? Otherwise I don't know what this could be. Looks like hardware fault to me.
Does the original firmware works?
And do you erased the esp before flashing awtrix light?

Benni620, 2023-08-25:
Yes, it is fully charged.
The Original Firmare looks good, I flashed it and the display showes time etc.

Yes I checked "erase" in the flasher

Blueforcer, 2023-08-25:
Ok first try to set the brightness via settings api to 50.
Also disable auto brightness.
Maybe the light sensor is broken

Benni620, 2023-08-25:
ABRI was already set to false, brightness to 255 did not bring any changes. But when I set ABRI to true, the pixels in the top row flash again and again. Occasionally the time also flashes briefly.
Very strange...
What would be a normal value for the brightness? I have on a cloudy day "lux": 24, in the house.
Thank you very much for your effort!
[image]

Benni620, 2023-08-25:
So it's really very funny.

I have now noticed that the problem only occurs when my laptop is on. When I turn it off, the display works. I will search the next few days a bit around what that can be, should I find something I write again.

Blueforcer, 2023-08-25:
That's indeed crazy 🤣

Blueforcer, 2023-08-29:
i will close this issue, since it looks like a hardware issue

closed 2023-08-29 by Blueforcer