Blueforcer/awtrix3 #276: [BUG] Single app still causes animation

Empty2k12, 2023-08-26:
# Bug report

## Describe the bug

I sometimes have a single app running at a time, and I am using a transition animation. With the single app, the transition still plays.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - AWTRIX-LIGHT version: 0.81

## To Reproduce

Delete all apps but time

## Expected behavior

No animation plays when the app "changes" since the pixels do not change

## Screenshots

[image]

Blueforcer, 2023-08-26:
I need more informations, I can't reproduce it.
What's apps do you had before (native or custom), how you delete them?

Empty2k12, 2023-08-26:
Hello @Blueforcer, I installed Awtrix and then deleted the default apps manually by sending `{"BAT": false}`, `{"TEMP": false}`, `{"HUM": false}`, `{"DAT": false}` to `MYTOPIC/notify` via MQTT and then rebooted via the UI at the local IP of my clock.

Empty2k12, 2023-08-27:
I have been able to reproduce the bug. I am sending a short-lived app: `{"text": "test", lifetime: 5, pushIcon: 2}`, after it expires, the bug happens.

Basically, I am trying to display a new sensor value as it gets recorded, but only once. Therefore, the short lifetime. However, both sending as an app and as a notification have some drawbacks: Sending as a notification, will queue sensor values, meaning they are shown in sequence, which is good. However, notifications don't seem to animate in like apps do. Sending as an app will display the animation, but it seems like the lifetime now also runs in sequence, meaning only one of two sensor values will show until both apps run out of their lifetime.

Which is the best way for my scenario: Short lived notification but with an animation?

Blueforcer, 2023-08-29:
Should be fixed in v0.82