Blueforcer/awtrix3 #278: [FEATURE REQUEST] Lower minimum brightness on auto brightness

claell, 2023-08-27:
# Feature Request

### Is your feature request related to a problem?

With auto brightness on, the brightness in a dark room is still too high for me. Not sure, what the actual brightness percentage was, though. So possibly, it already was at 1 % and that is just still too bright.

### Describe the solution / feature you'd like

Lower minimum brightness

### Describe alternatives you've considered

Reading brightness in and based on that, turn off the display or enable some night mode via the API.

### Additional context

None

Blueforcer, 2023-08-27:
https://blueforcer.github.io/awtrix-light/#/dev

closed 2023-08-27 by Blueforcer

claell, 2023-08-27:
Thanks!

> Create a dev.json in your filemanager.

That refers to the AWTRIX light file manager?

claell, 2023-08-27:
And in case, that's still too bright: Is that a hardware limitation? Or would it be possible to dim the matrix even further, and is just not supported in AWTRIX Light?

Blueforcer, 2023-08-27:
Its not possible.