Blueforcer/awtrix3 #279: [FEATURE REQUEST] Access device stats through HTTP API

claell, 2023-08-27:
# Feature Request

### Is your feature request related to a problem?

Right now, I am using the HTTP API (not MQTT) for some automation with AWTRIX light. Unfortunately, it currently doesn't seem to give me device values like temperature, brightness sensor, etc. through that way.

### Describe the solution / feature you'd like

Either one URL to get all values at once, or one to request them separately.

### Describe alternatives you've considered

As the values currently seem only to be exposed through MQTT, one could probably try to get them from there. That would unfortunately require quite some overhead.

### Additional context

None

Blueforcer, 2023-08-27:
did you read the documentation?
Its the very first entry...
https://blueforcer.github.io/awtrix-light/#/api?id=status

claell, 2023-08-27:
I thought I read it. Sorry, must have skipped that part.

closed 2023-08-27 by claell