Blueforcer/awtrix3 #283: [BUG] Icon GIFs 0B in filesystem

stnokott, 2023-08-30:
# Bug report

## Describe the bug

When downloading icons via the Web Interface, some downloaded files are 0B on the filesystem.
[image]
This results in the icon being displayed as black when used on the clock.

I could not 100% determine which icons are affected and which ones aren't. Icon 87 for example is also animated, but does work.

Alternative approaches I tried but resulted in the same issue:
- Download the GIF to my PC, then upload to filesystem manually
- Download the GIF to my PC, upload to ezgif.com/maker, download from there, upload to filesystem manually (as mentioned in https://github.com/Blueforcer/awtrix-light/issues/263#issuecomment-1682236726)
- Download the GIF to my PC, replace all transparent pixels with black, then upload to filesystem manually
- Download the GIF to my PC, remove the alpha channel, then upload to filesystem manually

I'm at my wits end and would love to know if I'm missing something obvious or if this is really a bug...

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: v0.81

## To Reproduce

1. Open "Icons" tab
2. Type in ID 49986
3. Click "Preview" -> looks fine
4. Click "Download" -> notification "Icon saved"
5. Open filesystem, observe how the icon file is 0B
6. Try to add to custom app, observe how the icon is only black pixels

## Expected behavior

The regular approach, but also at least one of the other approaches I tried should have resulted in a non-empty icon file.

## Screenshots

[image]
The icon being properly *previewed* in the Web UI.

## Logs

Might be helpful, but I don't think the firmware itself has a log...

## Additional context

-

Blueforcer, 2023-08-30:
Have a look at the free space in the right top corner of the filebrowser.
If it's full, all new files will be 0bytes.
That's a unchangeable behavior of the ESPs filesystem

stnokott, 2023-08-30:
Oh wow that was a quick reply.
Yeah the bar is red. Need to admit, I was in denial, because I only have three icons uploaded, so I didn't think space would become an issue...

Custom apps only take RAM, no flash space, correct?
Is it normal that my flash space is full with only 3 icons?

Blueforcer, 2023-08-30:
hover you mouse over the red bar. It tells you how much is the total flashsize.
Tell me this please

stnokott, 2023-08-30:
60KiB 😬

stnokott, 2023-08-30:
Not sure if that's relevant, but my config.json is already taking 11.37KiB too.

Blueforcer, 2023-08-30:
You should have a total of 256kb.
I assume that you forgot to check "erase" at your first flashing via webflasher..

stnokott, 2023-08-30:
Based on your replies, I did a complete wipe and re-flash and now I have 256kb.
I swear I erased when I flashed, but here we are.

Thanks for the swift replies, really appreciated!

closed 2023-08-30 by stnokott

Blueforcer, 2023-08-30:
You're welcome. 
Next release needs also a complete reflash with wipe, then everyone has 1,3MB free of flash space

stnokott, 2023-08-30:
> then everyone has 1,3MB free of flash space

Very excited for that!