Blueforcer/awtrix3 #284: [FEATURE REQUEST] Autostart as soon as power is supplied

borsTiHD, 2023-08-30:
# Feature Request

### Is your feature request related to a problem?

I have my clock connected to the PC via USB, which causes the clock to go off at some point during the night. As soon as I switch on the PC again, the clock is connected to the power again, but then I have to switch it on manually every time.

### Describe the solution / feature you'd like

I would like to see an optional setting to automatically boot the clock when it gets power.

### Describe alternatives you've considered

Another idea would be a kind of deep sleep mode, into which the watch falls as soon as it has less than 10-20% battery. As soon as power is restored or an API call is made, the watch could wake up.

### Additional context

Since my clock gets its apps from a small server app via API requests, it only displays current data when my PC is turned on. That would be one of the reasons I don't have them on permanent power.
I hope the idea finds approval. Regardless, I would like to say how great I find the project and it's fun to tinker with it. The new app I also find great, hope it gets more features. :)

Blueforcer, 2023-08-30:
First of all, the battery in this device is bullshit and looses it power in short lifetime. So in half of a year or before, it will kick in deep sleep very fast. (Besides the excessive rebuild of the firmware)

Automatic power on isn't possible because the ESP doesn't have any connection to the power management. The ESP has no power until you turn it on with both keys.

I recomment to turn off the matrix via API as soon you shutdown your PC. This saves lots of power and maybe will survive a day.

closed 2023-08-30 by Blueforcer

borsTiHD, 2023-08-30:
Sad to hear, but understandable.
Thanks for your response and I will try to switch off the matrix via API, thanks for the idea. 👍