Blueforcer/awtrix3 #288: [BUG] MQTT options not showing anymore

LucaZiegler, 2023-08-31:
# Bug report

## Describe the bug

Add a description of the bug. Detail the expected behaviour in contrast with the behaviour you're observing.

## Additional information

- Devices involved:
  - Model: Ulanzi TC001
  - AWTRIX-LIGHT version: 0.83

## To Reproduce

Steps to reproduce the behavior:
- The mqtt configuration tab is not showing anymore after an update and homeassistant is not finding the ulanzi anymore

## Expected behavior

- Able the change the mqtt configuration

## Screenshots
[image]

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.

Blueforcer, 2023-08-31:
Check the config.json. maybe it's broken. You can delete this file and reboot, awtrix will generate a new one.

LucaZiegler, 2023-08-31:
> All that stuff for the Webinterface and the entered settings.

Where can I get a Template?
Its not mentioned in the Documentation..

Blueforcer, 2023-08-31:
As I told you, just delete this file and reboot. Awtrix will generate a new one.

LucaZiegler, 2023-08-31:
> As I told you, just delete this file and reboot. Awtrix will generate a new one.

Ah sorry, yeah that worked, thx!

closed 2023-08-31 by LucaZiegler