Blueforcer/awtrix3 #290: [FEATURE REQUEST] App staleness overlay

kodek, 2023-09-02:
# Feature Request

### Is your feature request related to a problem?

Yes. I am sending app refreshes every 5 minutes, but my backend server is not always available, or some of the apps may fail to update. When the backend stops sending periodic app refreshes, I end up with stale pages (e.g. “sunny” weather at night 😄).

**I _did_ just notice that there’s a `lifetime` parameter.** This does solve my issue of displaying incorrect information. However, I would like these reporting failures to draw more attention in the clock.

### Describe the solution / feature you'd like

When an app has not been updated for longer than `[duration]`, **decorate the app with an overlay that indicates staleness**.

Overlay ideas:

 - Subtle pulsing red dot(s) on the left side of the screen (does this conflict with other indicators?).
 - Pulsing red dashes along the bottom — similar to week day indicators.
 - 1px red border around the entire app, maybe with a soft glow. (Pulsing might be too harsh and distracting.)
 - Blink a “question mark” icon on top of the app icon (see LaMetric **icon IDs #55064 and #55065**).
 - Replace text with something like “<App>: no recent update”

#### API changes

The API can be implemented in one of two ways:

1. Use `lifetime` along with a new parameter (e.g. `lifetimeMode` string):

This could allow different behaviors:

  a. `close` (Default): the app disappears.
  b. `markStale`: the app applies one of the effects above.
  
  (Why not a boolean? Adding a third mode would require breaking changes)

2. Introduce a new `watchdogDuration` or `stalenessDeadline` integer parameter.

This allows both `lifetime` and the new parameter to co-exist. It would allow apps to report staleness for some duration, then disappear if they become _too_ stale.

Blueforcer, 2023-09-04:
You can test it here:
https://discord.com/channels/546407049148366859/1143984083236569148/1148290038673723515

kodek, 2023-09-04:
> You can test it here: https://discord.com/channels/546407049148366859/1143984083236569148/1148290038673723515

Thanks! Just tried it out, and it looks good.

One potential bug: the overlay is not present until the app leaves the foreground and comes back. Is it possible to force a refresh?

Blueforcer, 2023-09-04:
No, thats how the scaffold works. Awtrix only checks lifetime during the app switches.

closed 2023-09-06 by Blueforcer