Blueforcer/awtrix3 #291: No IconTab in Webinterface[BUG]

DonatPlayZ, 2023-09-03:
# Bug report

## Describe the bug

Hello, since Today i got nomore "Icon" tab in the Webinterface. I already tried to "reupdate" to v. 0.83 still no tab for the icons, yesterday everything worked and i uploaded some icons

## Additional information

- Devices involved:
  - Model: Ulanzi Awtrix Smart Pixel Clock 2882 (TC001)
  - AWTRIX-LIGHT version: 0.83

## To Reproduce

Steps to reproduce the behavior:

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Logs

```text
(optional) Add relevant logs which could help tackle the problem.
```

## Additional context

Add any other context about the problem here.
[image]

Blueforcer, 2023-09-03:
Did you modified the config.json?

Delete ist and reboot awtrix to create a new one

closed 2023-09-03 by Blueforcer

robson80, 2023-11-15:
Hi,
I just had the same issue with version 0.90. I changed nothing, but uploaded 5 icon files in total. 

Also, some other tabs were not visible anymore: "Time", "Auth".
Furthermore, in "MQTT" tab only the "Broker" field was visible. All other fields (e.g. "Port", "Username") were gone.

As @Christian-4711 already stated, deleting file "DoNotTouch.json" and restaring the ESP helped.
What an irony when looking at the file name ;)

manalishi70, 2024-01-19:
Same behavior here: AWTRIX Light 0.94