Blueforcer/awtrix3 #292: [BUG] unstable WiFi

martz170, 2023-09-04:
# Bug report

## Describe the bug

Hello, i have problems with 2 Ulanzi Clocks. After 2~3 Days both clocks are loosing Wifi. No IP address and both 
Pixels are blinking ( red and yellow ). After restart of clock all is good again.
Clocks are receiving IP with reserved IP from DHCP. Other devices are working fine, so no Network issue here.

Firmware is 0.83

Please provide Infomation to grab logs here if possible.

Thank you.

Blueforcer, 2023-09-04:
please check the issues before opening a new one:
https://github.com/Blueforcer/awtrix-light/issues/197
https://github.com/Blueforcer/awtrix-light/issues/289
https://github.com/Blueforcer/awtrix-light/issues/254

closed 2023-09-04 by Blueforcer

martz170, 2023-09-04:
Hello Stephan,

thank you for pointing me the to the threads, but this really didn´t help.
I have also Unify AP´s but no channel optimisation and fixed Channels.
Also have a lot of IOT devices like WLED on ESP-32 and Tasmota etc. and all this devices are in the same IOT WLAN and have no problems whith Channel hopping or reconnection after AP reboot etc.
 
What i found is that Awtrix has problems with reconnection after AP reboot. Also reboot from Awtrix without the configured wlan is reachable - Awtrix then goes to AppMode which is not really useful.
Have a look on Tasmota - they are try to reconnect without start the app mode. Also WLED is really stable. No problem here after Access Point is rebooting or channel change. Since WLED is working also on ESP32 i see that the Hardware is capable to do this.

Please have a look at you code. 

Thank you.