Blueforcer/awtrix3 #294: Automatic shutdown TC 001 ULANCI

DichterKajo, 2023-09-04:
Hello all,

awtrix light is very great, but, the clock is power off after a few hours without any reasons.
The watch must start with the two buttons, no sleep mode / standby.

Actuall, i use the clock with the firmware 0.83 and the iOBroker app "awtrix-light" from Matthias Kleine.

If there any LOG-File, who gave me a little more information what happens?

Sorry for my bad english

Greetings Kajo

Blueforcer, 2023-09-04:
AWTRIX is not able to shutdown the clock. this is not possible hardware-wise.
Sounds like you doesnt provide enough power and the battery is discharging over time.
Because of the poor hardwaredesign the clock shutsdown when the battery is empty , even with USB connected

DichterKajo, 2023-09-04:
Ok, i have demount the batterie, thats right. 
I will check this with an another Clock, who the batterie is original mount, with the same charger.

closed 2023-09-04 by Blueforcer

simonszu, 2024-01-03:
I have the same issue. I have connected the clock to a power brick and a proper USB-C cable which is able to charge my Macbook. The battery level which is displayed went up from 46% to 80% when i last checked, but the clock shutdown nevertheless. The green light next to the charging port is green at all times. 
After restarting the clock the battery is shown with 0%.

@trip5 I read your linked issue, but my clock is only like 4 days old (ordered it on second christmas day). I really do not want to disassemble it. Anything else i could check?

trip5, 2024-01-04:
You may be overcharging it?  Try using a 2A charger.

simonszu, 2024-01-06:
Unfortunately no. I used a regular 5V 3A charger, and the clock did not stay on longer than 6 hours. 

What is interesting: when i try to start it by pressing left and right simultaneously, the first time nothing happens, not even the beep. Only after releasing and pressing both buttons again, the clock beeps and boots. 

Since the clock apparently seem to have a state after its display switches off after couple of hours, i think it is still somehow running, otherwise if it would be completely switched off, it would boot up at the first button press. 

But i have no further idea. Maybe a constant serial monitor until the clock "switches off". Does the clock send periodic messages via serial when powered on?

Blueforcer, 2024-01-06:
> What is interesting: when i try to start it by pressing left and right simultaneously, the first time nothing happens, not even the beep. Only after releasing and pressing both buttons again, the clock beeps and boots. 

That's a information we can work with.
Is the Webinterface accessable? If so,  only the matrix was turned off by API.

simonszu, 2024-01-07:
Hm, indeed. Apparently the clock was automatically recognized as a light by home assistant and some automations were firing.

cbirkenbeul, 2024-03-02:
Oh my gosh. Thanks @simonszu for this reply. That was exactly my problem too. I have an automation with poweroff all lights at night and my AWTRIX is also powered off by this automation.