Blueforcer/awtrix3 #295: unstable WiFI Reopened #292

martz170, 2023-09-04:
Hello Stephan,

thank you for pointing me the to the threads, but this really didn´t help.
I have also Unify AP´s but no channel optimisation and fixed Channels.
Also have a lot of IOT devices like WLED on ESP-32 and Tasmota etc. and all this devices are in the same IOT WLAN and have no problems whith Channel hopping or reconnection after AP reboot etc.
 
What i found is that Awtrix has problems with reconnection after AP reboot. Also reboot from Awtrix without the configured wlan is reachable - Awtrix then goes to AppMode which is not really useful.
Have a look on Tasmota - they are try to reconnect without start the app mode. Also WLED is really stable. No problem here after Access Point is rebooting or channel change. Since WLED is working also on ESP32 i see that the Hardware is capable to do this.

Please have a look at you code. 

Thank you.

_Originally posted by @martz170 in https://github.com/Blueforcer/awtrix-light/issues/292#issuecomment-1705306962_

Blueforcer, 2023-09-04:
I am not a friend of comparing myself or others with other projects. Everyone has their own ideas.
But if you know why awtrix doesn't reconnect (I can't test it, because it works fine for me, as it does for 99% of other users) then you are free to open a PR.

martz170, 2023-09-04:
Hi,

as long as other projects will not have this issues why not looking how they  do it ??

Anyway, i am not an Programmer, but an IT guy working on Servers and Networks. I can test your firmware in differnt conditions -no problem, but i can´t give code.

Since this problem concern many People ( see #197 ) i will be worth to work on this.

I searched a bit aund found this:
https://randomnerdtutorials.com/solved-reconnect-esp32-to-wifi/

Maybe it helps.

If there is something test - count on me. Please don´t close this issue too fast. Maybe others has also interesting links.

Thank you

closed 2023-09-05 by Blueforcer