Blueforcer/awtrix3 #296: [FEATURE REQUEST]

DrRSatzteil, 2023-09-05:
# Feature Request

### Is your feature request related to a problem?

I'd like to keep the state of the display (On/Off) in sync with my smart home solution. However the display state cannot be retrieved via mqtt/http.

### Describe the solution / feature you'd like

Add the current display state to the stat topic/endpoint.

### Describe alternatives you've considered

Just live with it 😀 

### Additional context

If this already exists please let me know ☺️

Blueforcer, 2023-09-05:
You can get it with the MATP key in the settings api

DrRSatzteil, 2023-09-05:
Wow I completely missed this endpoint 🙈

Sorry for opening this issue but thanks for your fast feedback! Issue closed.

DrRSatzteil, 2023-09-05:
Sorry to reopen this but is there a way to acutally receive settings via mqtt? It seems that this is only possible through the http api, right?

Blueforcer, 2023-09-05:
Its not possible, mqtt is not made for requests.

DrRSatzteil, 2023-09-05:
Would it be an option for you to have a sendsettings topic like for the screen/sendscreen pair? I think this would be a great compromise between sending in intervals and not being able to access the information at all via mqtt.

DanielHabenicht, 2023-10-14:
One more question regarding the `MATP` key: Should it be added to the `api.md` table? 
Because one only gets to know that it exists after trying out or finding this issue. 
It probably might also be a good idea to add the `GET` Option to the settings table.

Blueforcer, 2023-10-14:
No, that was just a key I needed for the app.

It will be available in the stats API next release

closed 2023-10-15 by Blueforcer

DanielHabenicht, 2023-10-15:
Sorry to bother you again! 
It was already available in the stats API (for version 0.88), it just was not documented in the markdown docs (and still isn't). Should I add a PR to document it?

By the way, are you interested in hosting an [openapi](https://swagger.io/specification/) specification for the API? 
This should make it easier to integrate with any 3rd party systems. I almost have it done for an Integration with HA I am building, so I am happy to contribute back.